PX4/PX4-SITL_gazebo-classic #726: Drone pitches when pitch command is sent to gimbal on typhoon h480

nilesh-aggarwal, 2021-03-09:
Hi, I need the gimbal on typhoon-h480 to point downwards. I followed this (https://docs.px4.io/master/en/advanced/gimbal_control.html) and decided on sending mavlink messages using .plan files on qgroundcontrol. 

Using this .plan file

####
{
    "fileType": "Plan",
    "geoFence": {
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 15,
        "firmwareType": 12,
        "hoverSpeed": 5,
        "items": [
            {
                "autoContinue": true,
                "command": 22,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    15,
                    0,
                    0,
                    null,
                    47.3980598,
                    8.5456161,
                    6
                ],
                "type": "SimpleItem"
            },

            {
                "autoContinue": true,
                "command": 204,
                "doJumpId": 3,
                "frame": 2,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    0,
                    0,
		            0

                ],
                "type": "SimpleItem"
            },

            {
                "autoContinue": true,
                "command": 205,
                "doJumpId": 3,
                "frame": 2,
                "params": [
                    90,
                    90,
                    5,
                    5,
                    0,
                    0,
                    0

                ],
                "type": "SimpleItem"
            }
            
        ],
        "plannedHomePosition": [
            47.3977418,
            8.5455937,
            488.184
        ],
        "vehicleType": 2,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 1
    },
    "version": 1
}

###

This should takeoff  the drone to 6 meters of height then using (MAV_CMD_DO_MOUNT_CONFIGURE) set stabilize pitch, roll, yaw to FALSE, then using (MAV_CMD_DO_MOUNT_CONTROL) set pitch and roll to 90 deg, But when this command is sent the drone starts to pitch and move forward.

My end goal is to make the camera on the typhoon h480 downfacing.

Jaeyoung-Lim, 2021-03-09:
@nilesh-aggarwal The `.plan` files are NOT used for sending mavlink messages from QGroundControl.

The `.plan` file is loaded on the vehicle when you upload the mission and is read on the firmware side. Therefore your mission file does not have a direct connection to the mavlink messages.

For example, if you close your QGroundcontrol in the middle of the mission, the vehicle will still be able to follow the mission

nilesh-aggarwal, 2021-03-09:
@Jaeyoung-Lim
Then how should I send mavlink messages? 
Thanks in advance

closed 2021-03-14 by LorenzMeier