PX4/PX4-SITL_gazebo-classic #729: Racer drone model or edititing gazebo_motor_model plugin

iftahnaf, 2021-03-15:
Hi all,

I have some questions regarding racer drone model in PX4-sitl_gazebo.

1) Is there a racer model for gazebo-px4 SITL? I couldn't find anything like this online.
By racer, I mean low hover thrust (about 15-20%) and a maximum speed of ~35 [m/s]. 

2) Alternatively, I tried to change some of the iris's `motor_model` plugin. After increasing the `motorConstant` parameter I was able to hover at 21% thrust(instead of 55%), but I couldn't find any parameter which causes the maximum velocity to be greater than 23 [m/s] (including `rotorDragCoefficient`). Am I missing something there? 

Thanks in advance,
Iftach.

Jaeyoung-Lim, 2021-03-15:
> Is there a racer model for gazebo-px4 SITL? I couldn't find anything like this online.
By racer, I mean low hover thrust (about 15-20%) and a maximum speed of ~35 [m/s].

No there is no racer model. If you can contribute one, that would be great

> Alternatively, I tried to change some of the iris's motor_model plugin. After increasing the motorConstant parameter I was able to hover at 21% thrust(instead of 55%), but I couldn't find any [...]

If you say maximum velocity I believe this is in mission mode: This would probably also include tuning the firmware.

iftahnaf, 2021-03-15:
Hi @Jaeyoung-Lim, Thanks for the answer.

For sure, I'll start working on it in the next few weeks.

Actually, I'm on v.10 firmware with `OFFBOARD` mode. After changing `motorConstant` parameter I tuned the PID gains. I sent acceleration commands (for simplicity just on Z direction). I also tried thrust command on attitude setpoint, and I monitor the velocity, but It doesn't go over 23 [m/s]. Is there something else I should consider changing?

Jaeyoung-Lim, 2021-03-15:
@iftahnaf The motor model takes account the inflow velocity, so you will not generate any thrust if you exceed that speed: https://github.com/PX4/PX4-SITL_gazebo/blob/6b6f4749a74215b9a4fb81cfaa99a3950f13464d/src/gazebo_motor_model.cpp#L221

bkueng, 2021-03-16:
@iftahnaf there's a number of PX4 parameter that you need to increase as well: MC_PITCHRATE_MAX (+other axis), MPC_MANTHR_MIN=0, MPC_MAN_TILT_MAX, MPC_TILTMAX_AIR, MPC_ACC_HOR_MAX, MPC_ACC_HOR, MPC_VEL_MANUAL, MPC_XY_VEL_MAX (not sure which ones applies to offboard), enable MC_AIRMODE.

iftahnaf, 2021-03-16:
> @iftahnaf The motor model takes account the inflow velocity, so you will not generate any thrust if you exceed that speed:
> 
> https://github.com/PX4/PX4-SITL_gazebo/blob/6b6f4749a74215b9a4fb81cfaa99a3950f13464d/src/gazebo_motor_model.cpp#L221

Thanks, this was exactly the problem. I changed this parameter (the 25.0) to my desired maximum velocity and rebuild with `make clean` and `make px4_sitl gazebo`.

@bkueng Ok Ill check the right values for those parameters.

Thanks again to both of you, this will be useful when I'll work on the racer model.

closed 2021-03-16 by iftahnaf

sharaf-roboking, 2026-10-27:
Hello @iftahnaf 
Have you found any racing drone models capable of reaching 45 m/s, or have you built one yourself?