PX4/PX4-SITL_gazebo-classic #733: PX4 SITL gazebo fails on Ubuntu 20.04 with linker error on pthreads

ivogeorg, 2021-03-29:
Hello,

I am following the [ubuntu dev install](https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets) and [gazebo sim setup](https://docs.px4.io/master/en/simulation/gazebo.html#running-the-simulation) instructions with a fresh clone of PX4 and fresh update of Ubuntu 20.04. I am getting the following linker error on `pthreads` (contents of `./build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log`):
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/ivogeorg/git-repos/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_be2d8 && [1/2] Building C object CMakeFiles/cmTC_be2d8.dir/src.c.o
[2/2] Linking C executable cmTC_be2d8
FAILED: cmTC_be2d8 
: && /usr/bin/cc -DCMAKE_HAVE_LIBC_PTHREAD   CMakeFiles/cmTC_be2d8.dir/src.c.o  -o cmTC_be2d8   && :
/usr/bin/ld: CMakeFiles/cmTC_be2d8.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ivogeorg/git-repos/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_4d733 && [1/2] Building C object CMakeFiles/cmTC_4d733.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_4d733
FAILED: cmTC_4d733 
: && /usr/bin/cc -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTC_4d733.dir/CheckFunctionExists.c.o  -o cmTC_4d733  -lpthreads && :
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Also, I am getting numerous errors related to protocol buffers like
```
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:7,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/sensors/Sensor.hh:33,
                 from /usr/include/gazebo-11/gazebo/sensors/CameraSensor.hh:24,
                 from /usr/include/gazebo-11/gazebo/sensors/DepthCameraSensor.hh:23,
                 from /home/ivogeorg/git-repos/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp:23:
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/gazebo-11/gazebo/msgs/altimeter.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
```
and
```
In file included from /usr/include/gazebo-11/gazebo/msgs/MessageTypes.hh:136,
                 from /usr/include/gazebo-11/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-11/gazebo/sensors/Sensor.hh:33,
                 from /usr/include/gazebo-11/gazebo/sensors/CameraSensor.hh:24,
                 from /usr/include/gazebo-11/gazebo/sensors/DepthCameraSensor.hh:23,
                 from /home/ivogeorg/git-repos/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp:23:
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:141:8: error: ‘bool gazebo::msgs::WrenchStamped::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’ marked ‘final’, but is not virtual
  141 |   bool MergePartialFromCodedStream(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:143:8: error: ‘void gazebo::msgs::WrenchStamped::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’ marked ‘final’, but is not virtual
  143 |   void SerializeWithCachedSizes(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/msgs/wrench_stamped.pb.h:145:30: error: ‘google::protobuf::uint8* gazebo::msgs::WrenchStamped::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  145 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
It might be relevant that I am using the [Lambda Stack](https://lambdalabs.com/lambda-stack-deep-learning-software), also freshly updated.

I have seen closed issues of both kinds from 2019 and 2020 but nothing worked for me. Any pointers or help would be greatly appreciated!

Thanks in advance!

Jaeyoung-Lim, 2021-03-29:
@ivogeorg We are verifying builds in the CI for Ubuntu Focal(https://github.com/PX4/PX4-SITL_gazebo/runs/2168943276?check_suite_focus=true), so I don't think this is related to the code, but rather your system. 

I don't have any experience with the lambda stack - have you tried without it?

ivogeorg, 2021-04-04:
@Jaeyoung-Lim, thanks for your reply! Several things:

1. You are pointing me to an automated or scheduled test run which is based on a docker image from Nov 2020 [px4io/px4-dev-simulation-focal:2020-11-18](https://hub.docker.com/layers/px4io/px4-dev-simulation-focal/2020-11-18/images/sha256-fe7a9913ed9e52a94e0424b4301ac028ea48b0cb9750160042629f0b624ffb3e?context=explore). 
2. Okay, I installed `docker` (same versions as your test run) and built the same container using the commands shown [here](http://docs.px4.io/master/en/test_and_ci/docker.html#calling-docker-manually). This one works until `gazebo` throws a `libGL` error, essentially being unable to find the visualization stack:
   ```
   [818/819] cd /src/PX4-Autopilot/build/px4_sitl_default/tmp && /src/PX4-Autopilot/Tools/sitl_run.sh /src/PX4-Aut...4_sitl_default/bin/px4 none gazebo tailsitter none /src/PX4-Autopilot /src/PX4-Autopilot/build/px4_sitl_default
   SITL ARGS
   sitl_bin: /src/PX4-Autopilot/build/px4_sitl_default/bin/px4
   debugger: none
   program: gazebo
   model: tailsitter
   world: none
   src_path: /src/PX4-Autopilot
   build_path: /src/PX4-Autopilot/build/px4_sitl_default
   GAZEBO_PLUGIN_PATH :/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo
   GAZEBO_MODEL_PATH :/src/PX4-Autopilot/Tools/sitl_gazebo/models
   LD_LIBRARY_PATH :/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo
   empty world, setting empty.world as default
   Using: /src/PX4-Autopilot/Tools/sitl_gazebo/models/tailsitter/tailsitter.sdf
   libGL error: No matching fbConfigs or visuals found
   libGL error: failed to load driver: swrast
   X Error of failed request:  GLXBadContext
     Major opcode of failed request:  151 (GLX)
     Minor opcode of failed request:  6 (X_GLXIsDirect)
     Serial number of failed request:  38
     Current serial number in output stream:  37
   ```
    (My monitors are attached to 1 of 4 GPUs and the mentioned Lambda Stack provides and maintains the NVIDIA driver, CUDA, cuDNN, and several machine learning frameworks. The container isolates the environment from this stack.)
3. At this point, I have to either point the container to the driver and libraries or I have to rebuild the container image with Lambda Stack in it.
4. The first option will take experimentation, but at best it will be hard-coded and non-future-proof.
5. The problem with the second is that the `Latest` tag of the container in (1) above fails with the same `pthreads` problem as shown in my original message. There are several tags between the image that is being run in the test and `Latest`, but I haven't yet tried them. Any one of those images might contain the problem. _The problem is not in the image itself, but that the Dockerfile assembles a dev environment which changes with time. The specific versions of all libraries and compilers do not seem to be enforced tightly. So, at some point, the defaults that `docker` used to build the image in Nov 2020 have silently changed and no longer result in a successful installation._
6. So, if I start with the corresponding [Dockerfile](https://github.com/PX4/PX4-containers/blob/master/docker/Dockerfile_simulation-focal) to try to add the [Lambda Stack](https://github.com/lambdal/lambda-stack-dockerfiles/blob/master/Dockerfile.focal) to it, I am likely to inherit the problem until it's eliminated (either by downgrading some libraries and enforcing the c++XX standard compiler option) or the PX4 stack is updated to work with the latest versions.
7. I will start by building an image from the Dockerfile first and see if that works. If there is a `pthreads` error again, the only solution is to trace and fix it. _(If not, I will experiment building writing a Dockerfile which combines the Lambda Stack and the PX4 dev-and-sim stack in one container image, which I will contribute widely once I get it working. Advanced avoidance scenarios for drones for DAA compliance will require machine learning.)_

Two questions:
1. Are you running tests against newer (e.g. Latest) `docker` images for `px4-dev-***-focal`?     
2. Is there any easy way to run gazebo outside the container, with additional MavLINK port mappings, where it will see the visualization options or is that a fix that is worse than the problem?

Thanks in advance!

Jaeyoung-Lim, 2021-04-05:
> Are you running tests against newer (e.g. Latest) docker images for px4-dev-***-focal?
@ivogeorg I never use the `latest` tag, since you lose track of what is inside. You can check how the tests are run in https://github.com/PX4/PX4-SITL_gazebo/blob/6fc9ef05b2d2cca06eef01971d28bf96ba97e6bb/.github/workflows/firmware_build_test.yml#L21

I am not sure where your libGL problem is coming from, but I also use this setup inside and outside the container and it should all work.

> Is there any easy way to run gazebo outside the container, with additional MavLINK port mappings, where it will see the visualization options or is that a fix that is worse than the problem?

Yes, this is the standard setup (not docker), and you can find the instructions in http://docs.px4.io/master/en/simulation/gazebo.html

Jaeyoung-Lim, 2021-04-05:
> Okay, I installed docker (same versions as your test run) and built the same container using the commands shown here. This one works until gazebo throws a libGL error, essentially being unable to find the visualization stack:

This is because you are running the gazebo client, and it doesn't have access to a display. Probably you need to map xhost to the container to get it to work / or you can run with `HEADLESS=1` 

> Are you running tests against newer (e.g. Latest) docker images for px4-dev-***-focal?

No, but I would never do this since the latest container will be always different (e.g. my latest will be different from yours depending on when we test)

> Is there any easy way to run gazebo outside the container, with additional MavLINK port mappings, where it will see the visualization options or is that a fix that is worse than the problem?
Yes, this is the standard setup (not docker) and you can find the instructions in http://docs.px4.io/master/en/simulation/gazebo.html

ivogeorg, 2021-04-05:
@Jaeyoung-Lim, thanks for writing back.
> Yes, this is the standard setup (not docker) and you can find the instructions in http://docs.px4.io/master/en/simulation/gazebo.html

This is what I did in the first place, and got the errors that I reported at the beginning. You said it's my environment, which it is, but I am not versed with the project enough yet to know how to reconfigure. To temporarily isolate myself from these issues, I went the `docker` route, as I reported above. And, as I also said, I did use the `Dockerfile` to build my own local container image for `px4-dev-simulation-focal` and seem to have been more successful with it. It builds fine and the `make` targets seem to work, except that I am still probably lacking the proper xhost mapping. Interestingly, it's not giving me the `libGL` error but it is complaining about not having a protocol and not being able to open the display 0:
```
[818/819] cd /src/PX4-Autopilot/build/px4_sitl_default/tmp && /src/PX4-Autopilot/Tools/sitl_run.sh /src/PX4-Aut...ild/px4_sitl_default/bin/px4 none gazebo none none /src/PX4-Autopilot /src/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /src/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /src/PX4-Autopilot
build_path: /src/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/src/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /src/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
No protocol specified
SITL COMMAND: "/src/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/src/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/src/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /src/PX4-Autopilot/build/px4_sitl_default/etc -> /src/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-04-05/00_55_27.ulg
[... 50 lines cut ...]
INFO  INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] [ecl/EKF] reset position to GPSstarting GPS fusion
starting GPS fusion

INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative

PX4 Exiting...
pxh> Exiting NOW.
```

Isn't using the px4 development and gazebo simulation Dockerfile as I have described more or less equivalent to running on the "latest container"? In what sense do you mean that it "will be always different"?

If I am to go about fixing the "standard setup" (no container), how would you recommend I go about doing so? One error points to `protobuf` conflicts, one to c++ standard mismatch, and the `pthreads` not being found is just plain weird, not least because it looks like it is generated by some dummy test CPP file.

ivogeorg, 2021-04-05:
If I do `xhost +` before I run a `make` target, I get the `libGL` error. If I don't, I get the "no protocol" and "cannot open display 0" errors. So, my running hypothesis should be that in the former case it tries to reach out of the container and cannot find the visualization stack, whereas in the latter case it does not and just abandons the rendering altogether.

Jaeyoung-Lim, 2021-04-05:
> Isn't using the px4 development and gazebo simulation Dockerfile as I have described more or less equivalent to running on the "latest container"? In what sense do you mean that it "will be always different"?

It should be, but since the [PX4/containers](https://github.com/PX4/PX4-containers) are not being verified pre-release, it is possible that the `latest` tag is not a meaningful release. The latest "official" container would be the one being used in the PX4 Autopilot CI, which would be something like https://github.com/PX4/PX4-Autopilot/blob/8400f2c9bc8c5c545e27fe59f7cd689d4fb1a62b/.github/workflows/sitl_tests.yml#L23

> If I do xhost + before I run a make target, I get the libGL error. If I don't, I get the "no protocol" and "cannot open display 0" errors. So, my running hypothesis should be that in the former case [...]

Can you try something like
```
xhost local:root
docker run -it --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix <container_name>:<tag> /bin/bash
```
Note that this is more about running gazebo inside the container rather than something related to PX4 directly. 

> If I am to go about fixing the "standard setup" (no container), how would you recommend I go about doing so? One error points to protobuf conflicts, one to c++ standard mismatch, and the pthreads [...]

The protobuff conflict is something that I have seen when running with deep learning frameworks (e.g. tensorflow). SITL Gazebo requires a certain version of protobuff. This is why I think we need to separate running PX4 in 20.04 and setting up your setup(e.g. running with Lambda).

My take on this problem is that the problem is not related to Ubuntu 20.04 in anyway - I use it daily and a lot of my team members do.
- There seems to be something going on with your system in a local setup. I would try and isolate protubuff first.
- Your docker environment is running as expected. You are just struggling to run it with a display connected. If you run it with `HEADLESS=1`, this will prevent the make target from running the gzclient and not complain about rendering.

ivogeorg, 2021-04-05:
@Jaeyoung-Lim, thanks again. Things are starting to make sense.

> Note that this is more about running gazebo inside the container rather than something related to PX4 directly.

I realized that. I am starting to tell things apart (flight controller, SITL layer, simulator, containers). _(Thankfully, at this stage I don't have to also use ROS - yet more complexity.)_ My task is to make changes primarily in the SITL layer, but probably also on both sides of it, in PX4 and gazebo, so this is a good thing :)

> The protobuff conflict is something that I have seen when running with deep learning frameworks (e.g. tensorflow). SITL Gazebo requires a certain version of protobuff.

Undoubtedly, as `tf` makes heavy use of protocol buffers. It might even be the original project for which they emerged. Where would I look to see what the version required by SITL gazebo is?

> This is why I think we need to separate running PX4 in 20.04 and setting up your setup(e.g. running with Lambda).

I am not sure this is entirely possible, as I run my monitors directly off the NVIDIA GPU. [Lambda Labs](https://lambdalabs.com/) maintain the NVIDIA driver, as well as the rest of the machine learning stack (CUDA, cuDNN, TensorFlow, Torch, etc.) through their Lambda Stack, which is essentially a Debian package. _(Here is the [Dockerfile](https://github.com/lambdal/lambda-stack-dockerfiles/blob/master/Dockerfile.focal) they provide for having the stack install inside an Ubuntu container. They run mostly Ubuntu on their machines, though have started to offer Windows 10 lately.)_ 

Do you think that, instead, if I can combine the layers for Lambda Stack and PX4-SITL_gazebo in one Dockerfile/image/container, that would allow gazebo to render without any mapping across the container boundary? That might require figuring the `protobuf` version first in any case.

I am trying to imagine what the use cases are for running gazebo with `HEADLESS=1`. Server? I need eventually to show that a drone, say the tailsitter, can "fly" with my modifications. Maybe I can just work off the logs, but that would be a level of indirection which might be quite clumsy.

Anyway, I will try the things you suggest and see what happens. Thanks for your help!

Jaeyoung-Lim, 2021-04-05:
@Jaeyoung-Lim In any case, it would be great if we can get to the bottom of this so more people can use SITL with learning frameworks.

Just to get you moving, is having SITL and your lambda infrastructure on different containers not an option? I believe you will only be needing some format of image / video through a UDP port?

ivogeorg, 2021-04-05:
`CMakeLists.txt`, of course. Thanks, @Jaeyoung-Lim.

Honestly, I hadn't gotten as far as to just keep the two stack in two separate containers. That might just get me going. In the long run, the throughput of the control stack and simulation data streams will probably impose further constraints. Or not, in which case training and simulation can be decoupled and scaled separately in the cloud.

Jaeyoung-Lim, 2021-04-19:
@ivogeorg Any updates regarding this?

ivogeorg, 2021-04-20:
@Jaeyoung-Lim I am working on creating containers with the right config. If you want, you can close this issue. How would you prefer that I communicate back when I am done? That may include edits to the documentation and other additions. A pull request or an issue?

Jaeyoung-Lim, 2021-06-07:
@ivogeorg Please open a pull request or a new issue :grin: Any contribution is welcome!

closed 2021-06-07 by Jaeyoung-Lim