PX4/PX4-SITL_gazebo-classic #738: Issues with Linking libphysics_msgs.so

sandeshworld, 2021-04-18:
```
[17/109] Linking CXX shared library libphysics_msgs.so
FAILED: : && /usr/bin/c++  -fPIC -O2 -g -DNDEBUG   -shared -Wl,-soname,libphysics_msgs.so -o libphysics_msgs.so CMakeFiles/physics_msgs.dir/Wind.pb.cc.o -L/usr/lib/x86_64-linux-gnu/gazebo-9/plugins  -L/home/sandesh/Documents/DroneSimulationDownloads/Firmware/build/px4_sitl_default/build_gazebo /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath /usr/lib/x86_64-linux-gnu/libSimTKsimbody.so /usr/lib/x86_64-linux-gnu/libSimTKmath.so /usr/lib/x86_64-linux-gnu/libSimTKcommon.so /usr/lib/libblas.so /usr/lib/liblapack.so /usr/lib/libblas.so -lpthread -lrt -ldl -lm /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so /usr/lib/x86_64-linux-gnu/libgazebo_physics.so /usr/lib/x86_64-linux-gnu/libgazebo_ode.so /usr/lib/x86_64-linux-gnu/libgazebo_transport.so /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so /usr/lib/x86_64-linux-gnu/libgazebo_util.so /usr/lib/x86_64-linux-gnu/libgazebo_common.so /usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so /usr/lib/x86_64-linux-gnu/libgazebo_opcode.so /usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.so /usr/lib/x86_64-linux-gnu/libgazebo_ccd.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_program_options.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libboost_iostreams.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/lib/x86_64-linux-gnu/libprotobuf.so -lpthread /usr/lib/x86_64-linux-gnu/libsdformat.so /usr/lib/x86_64-linux-gnu/libOgreMain.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so -lpthread /usr/lib/x86_64-linux-gnu/libOgreTerrain.so /usr/lib/x86_64-linux-gnu/libOgrePaging.so /usr/lib/x86_64-linux-gnu/libignition-transport4.so.4.0.0 /usr/lib/x86_64-linux-gnu/libignition-msgs1.so.1.0.0 /usr/lib/x86_64-linux-gnu/libignition-common1.so.1.1.1 /usr/lib/x86_64-linux-gnu/libignition-fuel_tools1.so.1.2.0 /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.9 /usr/lib/x86_64-linux-gnu/libprotobuf.so /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so -lglib-2.0 -lgobject-2.0 -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath /usr/lib/x86_64-linux-gnu/libSimTKsimbody.so /usr/lib/x86_64-linux-gnu/libSimTKmath.so /usr/lib/x86_64-linux-gnu/libSimTKcommon.so /usr/lib/libblas.so /usr/lib/liblapack.so -lpthread -lrt -ldl -lm /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so /usr/lib/x86_64-linux-gnu/libgazebo_physics.so /usr/lib/x86_64-linux-gnu/libgazebo_ode.so /usr/lib/x86_64-linux-gnu/libgazebo_transport.so /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so /usr/lib/x86_64-linux-gnu/libgazebo_util.so /usr/lib/x86_64-linux-gnu/libgazebo_common.so /usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so /usr/lib/x86_64-linux-gnu/libgazebo_opcode.so /usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.so /usr/lib/x86_64-linux-gnu/libgazebo_ccd.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_program_options.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libboost_iostreams.so -lpthread /usr/lib/x86_64-linux-gnu/libsdformat.so /usr/lib/x86_64-linux-gnu/libOgreMain.so /usr/lib/x86_64-linux-gnu/libOgreTerrain.so /usr/lib/x86_64-linux-gnu/libOgrePaging.so /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so -lglib-2.0 -lgobject-2.0 /usr/lib/x86_64-linux-gnu/libprotobuf.so /usr/lib/x86_64-linux-gnu/libprotobuf.so /usr/lib/x86_64-linux-gnu/libignition-math4.so.4.0.0 /usr/lib/x86_64-linux-gnu/libuuid.so /usr/lib/x86_64-linux-gnu/libuuid.so /usr/local/lib/libswscale.a /usr/local/lib/libavdevice.a /usr/local/lib/libavdevice.a -lAVDEVICE_LIBRARY_bz2-NOTFOUND /usr/local/lib/libavfilter.a -lAVDEVICE_LIBRARY_bz2-NOTFOUND /usr/local/lib/libavfilter.a /usr/local/lib/libswscale.a /usr/local/lib/libavformat.a /usr/local/lib/libavformat.a -lAVFORMAT_LIBRARY_bz2-NOTFOUND /usr/local/lib/libavcodec.a /usr/local/lib/libavcodec.a /usr/lib/x86_64-linux-gnu/libxcb.so /usr/lib/x86_64-linux-gnu/libxcb-shm.so /usr/lib/x86_64-linux-gnu/libxcb-xfixes.so /usr/lib/x86_64-linux-gnu/libxcb-render.so /usr/lib/x86_64-linux-gnu/libxcb-shape.so /usr/lib/x86_64-linux-gnu/libxcb.so /usr/lib/x86_64-linux-gnu/libxcb-shm.so /usr/lib/x86_64-linux-gnu/libxcb-xfixes.so /usr/lib/x86_64-linux-gnu/libxcb-render.so /usr/lib/x86_64-linux-gnu/libxcb-shape.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libasound.so /usr/lib/x86_64-linux-gnu/libSDL.so /usr/lib/x86_64-linux-gnu/liblzma.so -lAVCODEC_LIBRARY_bz2-NOTFOUND /usr/lib/x86_64-linux-gnu/libz.so /usr/local/lib/libswresample.a /usr/local/lib/libavutil.a /usr/local/lib/libavutil.a -lm /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.9 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.9 -Wl,-rpath,/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/home/sandesh/Documents/DroneSimulationDownloads/Firmware/build/px4_sitl_default/build_gazebo && :
/usr/bin/ld: cannot find -lAVDEVICE_LIBRARY_bz2-NOTFOUND
/usr/bin/ld: cannot find -lAVDEVICE_LIBRARY_bz2-NOTFOUND
/usr/bin/ld: cannot find -lAVFORMAT_LIBRARY_bz2-NOTFOUND
/usr/bin/ld: cannot find -lAVCODEC_LIBRARY_bz2-NOTFOUND
collect2: error: ld returned 1 exit status
[17/109] Linking CXX shared library libnav_msgs.so
FAILED: : && /usr/bin/c++  -fPIC -O2 -g -DNDEBUG   -shared -Wl,-soname,libnav_msgs.so -o libnav_msgs.so CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o -L/usr/lib/x86_64-linux-gnu/gazebo-9/plugins  -L/home/sandesh/Documents/DroneSimulationDownloads/Firmware/build/px4_sitl_default/build_gazebo /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath /usr/lib/x86_64-linux-gnu/libSimTKsimbody.so /usr/lib/x86_64-linux-gnu/libSimTKmath.so /usr/lib/x86_64-linux-gnu/libSimTKcommon.so /usr/lib/libblas.so /usr/lib/liblapack.so /usr/lib/libblas.so -lpthread -lrt -ldl -lm /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so /usr/lib/x86_64-linux-gnu/libgazebo_physics.so /usr/lib/x86_64-linux-gnu/libgazebo_ode.so /usr/lib/x86_64-linux-gnu/libgazebo_transport.so /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so /usr/lib/x86_64-linux-gnu/libgazebo_util.so /usr/lib/x86_64-linux-gnu/libgazebo_common.so /usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so /usr/lib/x86_64-linux-gnu/libgazebo_opcode.so /usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.so /usr/lib/x86_64-linux-gnu/libgazebo_ccd.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_program_options.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libboost_iostreams.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so  [... cut ...]
FAILED: cd /home/sandesh/Documents/DroneSimulationDownloads/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/sandesh/Documents/DroneSimulationDownloads/Firmware/build/px4_sitl_default/build_gazebo -- -j2
[746/749] Linking CXX executable bin/px4
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

```

Unsure as to why I am getting this issue:
Ran with` v1.11.3`
and did `git submodule update --recursive`

Any help would be much appreciated

Jaeyoung-Lim, 2021-04-19:
@sandeshworld Did SITL ever work on your system? If so, you probably have some message incompatibility coming from the protobuff generated messages?

Try
```
make distclean
make submodulesupdate
make px4_sitl gazebo
```

sandeshworld, 2021-04-21:
Hey @Jaeyoung-Lim , I tried that and it still throws the same error. I am able to use

`make px4_sitl jmavsim` fine though. It's an issue with:

[2/97] Linking CXX shared library libnav_msgs.so
FAILED: : && /usr/bin/c++  -fPIC -O2 -g -DNDEBUG 

Thanks for your help!