PX4/PX4-SITL_gazebo-classic #739: GPS (compass?) unstable on the specific objects.

engcang, 2021-04-19:
Hi I am testing the multi robot systems using SITL.

But starting the SITL on the specific object (jackal, mobile robot) or landing on it make GPS (compass?) unstable, which makes the preflight failure because of compass number 1.
So I wonder what I am doing or using wrong.

I am using [this](https://github.com/PX4/PX4-SITL_gazebo/files/6335697/jackal_hanjeon.urdf.zip) urdf file for the mobile robot.
And I attached the [video](https://youtu.be/KEivxSLnDRw) for the issue.
As you can see, I manually controlled the iris to land and take off few times.
The GPS and compass are stable from the ground and even on the tank (from 2:02 and taking off again  at 2:19).

But right after the drone lands on the mobile robot (jackal), compass and GPS get unstable. (from 2:35)
Starting the SITL with iris placed on the same mobile robot resulted in the same issue and I do not understand why the drone continuously getting unstable GPS / compass.

Can you check over what might be the problem of the mobile robot's urdf file?? (I am using the default iris.sdf and also tested with customized models)

Thank you in advance.

Jaeyoung-Lim, 2021-04-19:
@engcang You can probably look at the logs and what exactly is the problem in EKF (if it is sensor data stopping, or just seeing more vibrations). Does this comment help?

engcang, 2021-04-20:
@Jaeyoung-Lim Thanks. Looking into the log file, the problem is the serious vibrations, which means this is 100% related to mobile robot's urdf file!
Let me fix from my side. thank you again!

closed 2021-04-20 by engcang

engcang, 2021-04-20:
For anyone who can also suffer this issue in the future,
add sufficient friction and correction parameters to the surface in `urdf` file where drone has to land as below:
~~~xml
  <mu1>10.0</mu1>
  <mu2>10.0</mu2>
  <kp>10000000.0</kp>
  <kd>100.0</kd>
  <minDepth>0.01</minDepth>
  <maxVel>1.0</maxVel>
~~~