PX4/PX4-SITL_gazebo-classic #742: Increasing /clock publish rate

iftahnaf, 2021-04-20:
Hi all,

Is there a way to increase `/clock` rate in px4 SITL gazebo?
Currently, it publish at 250 [Hz], I want to increase it to 1000 [Hz], is it possible? Couldn't find anything on the web about this issue.

Thanks,
Iftach.

TSC21, 2021-04-20:
This is not an issue: you can do it [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/worlds/yosemite.world#L45) for the `yosemite` world for example. Same applies to all the other worlds.

closed 2021-04-20 by TSC21

iftahnaf, 2021-04-20:
Hi @TSC21 , Thanks for the answer.

I tried to change this parameter, but I'm keep getting errors from the gazebo when launch the SITL:
`[gazebo-3] process has died [pid 7846, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode`

When I'm going back to `<real_time_update_rate>250</real_time_update_rate>` instead of `<real_time_update_rate>1000</real_time_update_rate>` everything works fine again.

Maybe I'm missing some other parameter that also need change?

iftahnaf, 2021-04-22:
Update:

I was able to get this work with the following commands:

on the `.world` file:
1. changing the `real_time_update` to 1000.
2. changing `max_step_size` to 0.001.

on `gazebo_mavlink_interface.cpp` :
3. changing `correct_real_time_update_rate` to 1000.
4. changing `correct_max_step_size` to 1000.

5. `make clean`
6. `make px4_sitl_default gazebo`

Thanks again!