PX4/PX4-SITL_gazebo-classic #744: upgrade the msgs from 'proto2' to 'proto3'

addy1997, 2021-04-28:
For my project, I need to upgrade the [msgs](https://github.com/PX4/PX4-SITL_gazebo/blob/master/msgs/Imu.proto) from proto2 to proto3. Is changing the syntax enough?

Jaeyoung-Lim, 2021-04-28:
@addy1997 I think whether if it is enough or not will depend on your project? Does this answer your question?

addy1997, 2021-04-28:
> @addy1997 I think whether if it is enough or not will depend on your project? Does this answer your question?

Sorry, I should have included more details. So, I am replicating the model of [this project](https://github.com/utiasSTARS/PhoenixDrone-sitl_gazebo) for my [project](https://github.com/addy1997/AlphaDrone-SITL_ignition-gazebo) based on AlphaDrones with Ignition gazebo that supports protobuf 3.

Jaeyoung-Lim, 2021-04-28:
@addy1997 I am not sure. Did you have any issues when updating?

addy1997, 2021-04-28:
> @addy1997 I am not sure. Did you have any issues when updating?

@Jaeyoung-Lim, thanks for your help.

closed 2021-04-28 by addy1997