PX4/PX4-SITL_gazebo-classic #745: Gazebo predicting downwards acceleration higher than g

manumerous, 2021-05-05:
## Introduction

While generating some logs using the `gazebo_standard_vtol` model i observed some strange and quirky behavior in gazebo. 
Sometimes the model would predict a downward acceleration higher than the one we would expect from gravity while the collective thrust from the 4 vertical actuators is more than sufficient to compensate gravity. 

## Data Collection

The following logs were obtained by manually controlling the vehicle with my Taranis X9D Plus RC. I tried, as good as manually possible to only change the throttle and none of the other inputs. 

I was using the following code:
`Commit: 472e191fef e2563388eb` of [https://github.com/ethz-asl/ethzasl_fw_px4](https://github.com/ethz-asl/ethzasl_fw_px4)

## Presented Data:
- The sensor_combined acceleration data is the measured accelerometer data in FRD body frame with positive z values indicating downward acceleration. This vector corresponds to the actual acceleration in FRD body frame minus the gravity vector (also in FRD of course). For this reason a value of 0 corresponds to a free fall.
- The actuator inputs as controlled by my RC. 
- The attitude quaternion. This excludes the possibility of the rotors pointing downwards or other strange behaviors. 

All three logs can be found [in my gdrive](https://drive.google.com/drive/folders/1gyR5u3quY27MbbO7KAJOn9Pa9v0BVT1T?usp=sharing) if you want to analyze them yourself. 

## Plots
### qpm_actuator_quirky.ulg
Here the simulation suddenly generates a quirky behavior as can be seen by the acceleration in z direction becoming positive, even though the collective thrust is far more than enough to compensate gravity. 
[image]

Furthermore, it seems that there were somehow additional pitch and roll actuator control commands generated, even though the right stick of my RC was not even touched. I am not sure though if that's something from the simulator or the PX4 Firmware. 
[image]

### 10_32_27.ulg
More of the same behavior can be seen in this example. Here the collective thrust and the vehicle attitude stay approx. constant while the acceleration in z direction does funky things. 
[image]

### qpm_gravity_test.ulg
Just to validate and show how the gravity should be displayed i wanted to include this plot. Here it can nicely be seen, that as expected the acceleration in z direction goes to 0 once all actuators are turned off. 
[image]

## Conclusion
This behavior seems very odd to me and I can not think of a physical effect, causing this drastic acceleration. For this reason i think it could be interesting to have a closer look at this since it could potentially be a bug. 

Let me know if you need any further information.

Jaeyoung-Lim, 2021-05-05:
@manumerous some comments
- I think you are assuming that your accelerometers are at your CG, which you would need to check if it is the case.
- You need to compare these with groundtruth accelerations. We dont know what you are looking at are coming from the sensors or the dynamics
- Your RC sticks are sending high level commands (depending on your flight mode). Therefore even if you dont touch the sticks the firmware will be always controlling a state- let it be position/velocity or even attitude.

manumerous, 2021-05-05:
Thanks for pointing this out. Yes it could indeed be that those measurements could come from a accelerometer placed outside the CG. I forgot to mention that I was flying in manual mode (attitude control mode)

At least the angular velocity and accelerations are quite large when this high downwards acceleration is generated:

[image]

Then i guess the question would be why does the firmware or simulator sometimes generate this really high angular accelerations. How could we check what kind of accelerations the simulator generates? Are they logged somewhere?

Jaeyoung-Lim, 2021-05-05:
@manumerous You can get it directly through the gazebo API (it gives you the acceleration of the origin)

manumerous, 2021-05-06:
In the `standard_vtol.sdf` file it seems to be mounted at the origin:

```
<link name='standard_vtol/imu_link'>
  <pose>0 0 0 0 0 0</pose>
  <inertial>
    <pose>0 0 0 0 0 0</pose>
    <mass>0.015</mass>
    <inertia>
      <ixx>1e-05</ixx>
      <ixy>0</ixy>
      <ixz>0</ixz>
      <iyy>1e-05</iyy>
      <iyz>0</iyz>
      <izz>1e-05</izz>
    </inertia>
  </inertial>
</link>
<joint name='standard_vtol/imu_joint' type='revolute'>
  <child>standard_vtol/imu_link</child>
  <parent>base_link</parent>
  <axis>
    <xyz>1 0 0</xyz>
    <limit>
      <lower>0</lower>
      <upper>0</upper>
      <effort>0</effort>
      <velocity>0</velocity>
    </limit>
    <dynamics>
      <spring_reference>0</spring_reference>
      <spring_stiffness>0</spring_stiffness>
    </dynamics>
    <use_parent_model_frame>1</use_parent_model_frame>
  </axis>
</joint>
```

Jaeyoung-Lim, 2021-05-06:
@manumerous Okay, then you need to check where the CG is compared to the origin

closed 2021-07-19 by Jaeyoung-Lim