PX4/PX4-SITL_gazebo-classic #755: Distance sensor lost error with typhoon_h480

Jaeyoung-Lim, 2021-05-15:
**Problem Description**
Distance sensor error is appearing when running the typhoon_h480 model
```
ERROR [ekf2] 3 - distance_sensor lost, generation 284 -> 286
ERROR [ekf2] 4 - distance_sensor lost, generation 284 -> 286
ERROR [ekf2] 5 - distance_sensor lost, generation 284 -> 286
ERROR [ekf2] 0 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 2 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 1 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 4 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 3 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 5 - distance_sensor lost, generation 336 -> 338
ERROR [ekf2] 0 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 2 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 3 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 5 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 1 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 4 - distance_sensor lost, generation 362 -> 364
ERROR [ekf2] 0 - distance_sensor lost, generation 440 -> 442
ERROR [ekf2] 3 - distance_sensor lost, generation 440 -> 442
ERROR [ekf2] 1 - distance_sensor lost, generation 440 -> 442
ERROR [ekf2] 4 - distance_sensor lost, generation 440 -> 442
ERROR [ekf2] 5 - distance_sensor lost, generation 440 -> 442
ERROR [ekf2] 0 - distance_sensor lost, generation 466 -> 468
ERROR [ekf2] 2 - distance_sensor lost, generation 466 -> 468
```

**Steps to reproduce**
```
make px4_sitl gazebo_typhoon_h480
```

Jaeyoung-Lim, 2021-05-31:
Seems like this was what introduced the failure https://github.com/PX4/PX4-Autopilot/pull/16636

JethroPhuah, 2021-05-31:
So are there any solutions to this problem yet? @Jaeyoung-Lim

TSC21, 2021-05-31:
@dagar

Jaeyoung-Lim, 2021-06-15:
@JethroPhuah This has been fixed by https://github.com/PX4/PX4-Autopilot/pull/17718

closed 2021-06-15 by Jaeyoung-Lim