PX4/PX4-SITL_gazebo-classic #757: motor_failure_plugin label:question

registera13, 2021-05-17:
Hello,

I have 2 questions regarding the motor_failure_plugin:

1) What is the command to publish to `/gazebo/motor_failure_num`? i have tried `pub = rospy.Publisher("/gazebo/motor_failure_num", String, queue_size=10)` and 
        `pub.publish("2")`
2)How to you enable the ros motor_failure plugin?

Jaeyoung-Lim, 2021-05-18:
Hi @registera13 , 

1) You need to create your own ROS node in order to publish to the rostopic. How to do this will well explained in this [link](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29)

2) You would need to add the plugin to the model in the sdf file. You can find more information [here](http://gazebosim.org/tutorials?tut=ros_gzplugins)

registera13, 2021-05-25:
Thanks Jaeyoung-LiM for the quick response,

i added the plugin to the iris.sdf, is this correct?
<plugin name="motor_failure" filename="libgazebo_motor_failure_plugin.so">
        <robotNamespace/>
        <ROSMotorNumSubTopic>/motor_failure/motor_number</ROSMotorNumSubTopic>
        <MotorFailureNumPubTopic>/gazebo/motor_failure_num</MotorFailureNumPubTopic>
</plugin> 

I was not able to spawn the vehicle when launching the gazebo. is there an example .sdf? not much documentation on how to use this plugin.

mariam0amin, 2024-05-16:
@registera13  were you able to solve this?