PX4/PX4-SITL_gazebo-classic #760: Error compiling PX4-SITL with RotorS package

jocacace, 2021-05-21:
Hi,

I am addressing a problem compiling the px4 sitl. The error occurs after I compiled the standalone RotorS library to perform some simulations without the px4 stack. The compilation error is reported in the following:
`gzserver: symbol lookup error: /home/jcacace/px4_fw_ws/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so: undefined symbol: _ZN8mav_msgs4msgs17CommandMotorSpeedC1Ev
`
Seems that the libgazebo_mavlink_interface.so plugin uses a different version of the CommandMotorSpeed message. Actually, I have seen that RotorS uses a different version of message, relying on gz_mav_msgs instead of mav_msgs. Sadly I need of both software on my system, do you think that is possible?

Jaeyoung-Lim, 2021-05-31:
@jocacace This package branched out of rotorS, therefore you will have a lot of duplicate packages if you use the two together. Why are you trying to use both?

Both packages have a plugin called mavlink_interface and uses a different implementation of messages and mavlink communication

jocacace, 2021-05-31:
Dear Jaeyoung-Lim,

Thank you for reply. The fact is that in some cases I develop low level controllers actuating the UAV considering the direct actuator control. I also find useful the Px4-sitl when I have to control the real system... so I need both. I am using two ROS workspaces to have both right now.

Jaeyoung-Lim, 2021-05-31:
@jocacace But you can still do direct actuator controls using this package. I am not sure what components of rotorS you are using that doesn't exist in this package.

closed 2021-06-04 by Jaeyoung-Lim

w407022008, 2024-05-28:
> Hi,
> 
> I am addressing a problem compiling the px4 sitl. The error occurs after I compiled the standalone RotorS library to perform some simulations without the px4 stack. The compilation error is reported in the following: `gzserver: symbol lookup error: /home/jcacace/px4_fw_ws/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so: undefined symbol: _ZN8mav_msgs4msgs17CommandMotorSpeedC1Ev ` Seems that the libgazebo_mavlink_interface.so plugin uses a different version of the CommandMotorSpeed message. Actually, I have seen that RotorS uses a different version of message, relying on gz_mav_msgs instead of mav_msgs. Sadly, I need of both software on my system working in ROS, do you think that is possible?
> 
> I am using Ubuntu 20.04 and ROS Noetic
> 
> Thanks,

Hi, there. It's valid if you just replace all "mav_msgs" in [CMakeLists.txt](https://github.com/ethz-asl/rotors_simulator/blob/cd813b7a8c375d677352aa20ad20047feb661126/rotors_gazebo_plugins/CMakeLists.txt#L214-L220) as any e.g mav_msg. The error is because of the conflict of libmav_msgs.so in both build.