PX4/PX4-SITL_gazebo-classic #761: Terrain following using typhoon_h480 model in Gazebo

dylantzx, 2021-05-27:
I am trying to implement terrain following using readings from a sonar sensor as the primary height source. However as a beginner, despite reading the PX4 guide, I am still quite lost as to how I can go about implementing this. 

Correct me if I am wrong, but this is how I think it would roughly work:
1) I would have to change some of the parameters in Qgroundcontrol, which are:
a. **EKF2_HGT_MODE** to range sensor
b. **MPC_ALT_MODE** to terrain following
2) I create another sonar sensor in the gazebo model sdf file (typhoon_h480) that publishes the readings to a rostopic via ROS, and if I can somehow feed the readings back to the PX4 as the primary height source, I would be able to implement the terrain following right?

However, I am not quite sure how to go about implementing the second part as I do not understand how I can get the PX4 to subscribe to my own sonar rostopic? Or is there already a rostopic that I can just publish the sonar readings to?

Just for more context, I followed the guide and have the PX4-Autopilot folder in my home directory. I am using a custom world that uses the **winding_valley_heightmap** and this is the code i use to start Gazebo, Mavros and px4:
```
cd ./PX4-Autopilot
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch vehicle:=typhoon_h480 world:=$(pwd)/../worlds/hills.world x:=-25.7 y:=27 z:=-1.2
```

Jaeyoung-Lim, 2021-05-27:
The sonar sensor in the typhoon h480 is facing forward for collision prevention. Did you move the sonar sensor to face downwards?

Jaeyoung-Lim, 2021-05-31:
@dylantzx Probably your problem is related to https://github.com/PX4/PX4-SITL_gazebo/issues/755

dylantzx, 2021-06-04:
Thank you. I will use the iris model with an added sonar plugin for now then.

closed 2021-06-04 by Jaeyoung-Lim