PX4/PX4-SITL_gazebo-classic #764: qusetion about  acceleration control

958117216, 2021-06-03:
Hi,I have some problems and hope to get your help.

To make the drone fly at the desire acceleration,my thought is make the drone at a  specified altitude and attitude.

For example, I tilt 5 degrees to produce acceleration of tan (5°)* g, but in the px4 gazebo simulation, it can only reach a certain speed and the acceleration is 0.Is there any resistance here I overlooked? Have you encountered this situation in simulation? Thanks and look forward to your reply.

Jaeyoung-Lim, 2021-06-04:
@958117216 Since the rotors have drag, your tilt angle does not necessarily result in acceleration. The vehicle will just reach the velocity in which the forward force and the drags are the same magnitude (also called as terminal velocity)

958117216, 2021-06-04:
@Jaeyoung-Lim  Thank you very much for your reply. 
How is the rotor drag calculated in gazebo? Is the following line of code? 
air_drag = -std::abs(real_motor_velocity) * rotor_drag_coefficient_ * body_velocity_perpendicular;

I just think the acceleration caused by tilt in the simulation is much smaller than the actual one.

Jaeyoung-Lim, 2021-06-04:
> I just think the acceleration caused by tilt in the simulation is much smaller than the actual one.

@958117216 I don't think this statement is necessarily true. This will depend on which airframe you are talking about, which is defined by several parameters. If you define your parameters close enough to your platform of interest the model in this repo behaves quite similarly with the real model except when you are doing extremely aggressive maneuvers. 

> How is the rotor drag calculated in gazebo? Is the following line of code?
> air_drag = -std::abs(real_motor_velocity) * rotor_drag_coefficient_ * body_velocity_perpendicular;

https://github.com/PX4/PX4-SITL_gazebo/blob/5c248898524cb670df496149de5e4692e16e69dc/src/gazebo_motor_model.cpp#L231

958117216, 2021-06-05:
@Jaeyoung-Lim  Thank you very much. I have another question. Does the drag produced by the rotor refer to air resistance + rotor resistance or just rotor resistance, because I noticed that the code contains  the  real_motor_velocity. 

I use the air resistance formula: F=(1/2)CρSV² to calculate the air resistance. I think air resistance is the only resistance during the flight of the UAV, which offsets the thrust generated by the tilt of the quadrotor magetan(Θ).

Sorry for not being able to introduce the background of the problem in detail. The flying mechanism is an x-type quadrotor and the motor is  Tmotor AIR2216KV880 with the T1045 propeller.

 I have changed some motor parameters according to https://github.com/PX4/PX4-SITL_gazebo/issues/110. 
The motor parameters are as follows:
      <timeConstantUp>0.0125</timeConstantUp>
      <timeConstantDown>0.025</timeConstantDown>
      <maxRotVelocity>1096</maxRotVelocity>
      <motorConstant>1.0555e-05</motorConstant>
      <momentConstant>0.06</momentConstant>    
      <commandSubTopic>/gazebo/command/motor_speed</commandSubTopic>
      <motorNumber>1</motorNumber>
      <rotorDragCoefficient>0.000806428</rotorDragCoefficient>
      <rollingMomentCoefficient>1e-06</rollingMomentCoefficient>

closed 2021-06-05 by 958117216