PX4/PX4-SITL_gazebo-classic #766: Follow me mode in Gazebo

dylantzx, 2021-06-04:
I am interested in trying out the PX4 follow me mode in gazebo, but I am required to publish the GPS position of the target using the FOLLOW_TARGET MAVLink message. However, I not quite sure how I can go about publishing the MAVLink message. 

I was thinking of attaching a GPS plugin to an object (for example a box), and be able to broadcast the location of the box to see how the drone moves to the location of the box. Would there be anyone who knows how I can go about implementing this?

Jaeyoung-Lim, 2021-06-04:
@dylantzx You can implement that by adding a plugin in gazebo: http://gazebosim.org/tutorials?tut=ros_gzplugins

dylantzx, 2021-06-07:
@Jaeyoung-Lim I did read the tutorial before but there wasn't a plugin for gps so I tried adding a gps plugin from the iris.sdf file to my box model in the .world file. However, I was not able to get the follow me function to work. I then tried using the ```roslaunch px4 multi_uav_mavros_sitl.launch``` and launched 2 drones with **drone 1** in follow me mode and **drone 2** in position mode flying around. However, I am still unable to get the drone 1 to follow drone 2. Is there something that I've missed out?

Jaeyoung-Lim, 2021-06-07:
@dylantzx Attaching a gps plugin to the box will only make gps location of the box published in gazebo transport. Which I suspect is not what you are trying to achieve.

> I then tried using the roslaunch px4 multi_uav_mavros_sitl.launch and launched 2 drones with drone 1 in follow me mode and drone 2 in position mode flying around. However, I am still unable to get the drone 1 to follow drone 2. Is there something that I've missed out?

I am not sure if the expected behavior is that the two drones will follow each other. They are independent systems, and are not sending `FOLLOW_TARGET` to each other. 

Reading your comments I am not sure if this even needs to be a gazebo plugin.
It appears to me that you only need to 
a. Acquire the gps location of the target (whether this is a box in gazebo or not)
b. Publish the `FOLLOW_TARGET` mavlink message to the autopilot in order to get the vehicle follow the target.

In order to achieve this, I think you can implement the functionality of a.b. as a gazebo plugin, or implement it externally for example using MAVSDK or ROS.

dylantzx, 2021-06-07:
Hello @Jaeyoung-Lim Thanks for the prompt reply!
I am currently trying to send the FOLLOW_TARGET mavlink message but am doing so with reference to the [First Application Tutorial (Hello Sky)](https://docs.px4.io/master/en/modules/hello_sky.html)

Following this tutorial, I should already be able to publish uORB messages to the FOLLOW_TARGET topic once I run the script with `pxh> follow_target` right? However, I am unable to get the drone to move to the intended location. It actually just flies left no matter what latitude and longitude I send and I am not quite sure why. Below is what I've written:

```
int follow_target_main(int argc, char *argv[])
{
	PX4_INFO("Sending timestamp,lat,lon & alt to follow_target uORB topic...");

	/* subscribe to vehicle_gps_position topic */
	int gps_sub_fd = orb_subscribe(ORB_ID(vehicle_gps_position));

	/* limit the update rate to 5 Hz */
	orb_set_interval(gps_sub_fd, 200);

	/* advertise follow_target topic */
	struct follow_target_s ft;
	memset(&ft, 0, sizeof(ft));
	orb_advert_t ft_pub = orb_advertise(ORB_ID(follow_target), &ft);

	px4_pollfd_struct_t fds[] = {
		{ .fd = gps_sub_fd,   .events = POLLIN },
	};

	int error_counter = 0;

	for (int i = 0; i < 100; i++) {
		/* wait for sensor update of 1 file descriptor for 1000 ms (1 second) */
		int poll_ret = px4_poll(fds, 1, 1000);

		/* handle the poll result */
		if (poll_ret == 0) {
			/* this means none of our providers is giving us data */
			PX4_ERR("Got no data within a second");

		} else if (poll_ret < 0) {
			/* this is seriously bad - should be an emergency */
			if (error_counter < 10 || error_counter % 50 == 0) {
				/* use a counter to prevent flooding (and slowing us down) */
				PX4_ERR("ERROR return value from poll(): %d", poll_ret);
			}

			error_counter++;

		} else {

			if (fds[0].revents & POLLIN) {
				/* obtained data for the first file descriptor */
				struct vehicle_gps_position_s gpsPos;
				/* copy sensors raw data into local buffer */
				orb_copy(ORB_ID(vehicle_gps_position), gps_sub_fd, &gpsPos);
				PX4_INFO("[Current] Lat: %8.4f\tLon: %8.4f\tAlt: %8.4f",
					 (double)gpsPos.lat,
					 (double)gpsPos.lon,
					 (double)gpsPos.alt);

				/* 
               				 Set a fixed lat,lon,alt for now to see if drone can fly to the position
               				 Starting Position while in hold mode:
               				 Lat: 381614783.0000	Lon: -1224546284.0000Alt: 496197.0000
              			*/
                		ft.timestamp = hrt_absolute_time();
                		ft.lat = 381610579.0f;
				ft.lon = -1224542632.0f;
				ft.alt = 492032.0f;

				orb_publish(ORB_ID(follow_target), ft_pub, &ft);
			}

		}
	}

	PX4_INFO("exiting");

	return 0;
}
```
This is the **final location** that the drone was at once the script has ended:
```Lat: 381621984.0000	Lon: -1224554984.0000   Alt: 496235.0000```

Jaeyoung-Lim, 2021-06-07:
@dylantzx I am not sure publishing uORB messages like that is expected to do what you are trying to acheive?

Have you looked at https://docs.px4.io/master/en/flight_modes/follow_me.html ?

dylantzx, 2021-06-07:
@Jaeyoung-Lim Hmmmmm...I thought it would work considering that I am publishing the FOLLOW TARGET message, and yes, I did look at that guide a few times already but there wasn't much information that could help me with simulating the follow me mode in gazebo. 

Here are the steps that I've done after launching:
1) Do a takeoff
2) Change from hold mode to follow me mode. 
3) Run the follow_target script

dylantzx, 2021-06-08:
@Jaeyoung-Lim I went to take a look at the code again and found this in the **follow_target.msg** file under `PX4-Autopilot/msg/follow_target.msg`
```
float64 lat			# target position (deg * 1e7)
float64 lon			# target position (deg * 1e7) 
```
I divided the lat and lon by 1e7 and fed it to the drone and was able to roughly move in the specified direction. Not sure if this is the way to do it though.

Jaeyoung-Lim, 2021-06-08:
@dylantzx If you are curious how the follow_target mavlink message is handled: https://github.com/PX4/PX4-Autopilot/blob/8b1d2b855115a3199e5aaf52a30618096f57ee3a/src/modules/mavlink/mavlink_receiver.cpp#L2279

dylantzx, 2021-06-08:
@Jaeyoung-Lim Thank you for the link. Is it correct for me to say that there is already a function to convert the MAVLink FOLLOW_TARGET messages into uORB messages and therefore, there shouldn't be a need for me to rewrite a function to send uORB messages but instead, I should try to get the MAVLink FOLLOW_TARGET messages?

I am currently referring to https://docs.px4.io/master/en/ros/mavros_custom_messages.html to try to subscribe to a GPS MAVROS topic (for example `uav1/mavros/gpsstatus/gps1/raw`) and send it as a MAVLink message. Hope I am in the right direction now? However, I am not too sure if I am doing it right?

```
#include <mavros/mavros_plugin.h>
#include <pluginlib/class_list_macros.h>
#include <iostream>
#include <mavros_msgs/GPSRAW.h>

 namespace mavros {
 namespace extra_plugins{

 class FollowTargetPlugin : public plugin::PluginBase {
 public:
     FollowTargetPlugin() : PluginBase(),
         nh("~gpsstatus/gps1")

    { };

     void initialize(UAS &uas_)
     {
         PluginBase::initialize(uas_);
         follow_target_sub = nh.subscribe("raw", 10, &FollowTargetPlugin::follow_target_cb, this);
     };

     Subscriptions get_subscriptions()
     {
         return {/* RX disabled */ };
     }

 private:
     ros::NodeHandle nh;
     ros::Subscriber follow_target_sub;

    void follow_target(
        int32_t lat, /*< [degE7] Latitude (WGS84) */
        int32_t lon, /*< [degE7] Longitude (WGS84) */
        float alt /*< [m] Altitude (MSL) */
    )
	{
		mavlink::common::msg::FOLLOW_TARGET ft = {};

		ft.lat = lat;
		ft.lon = lon;
		ft.alt = alt;

		UAS_FCU(m_uas)->send_message_ignore_drop(ft);
	}

    void follow_target_cb(const mavros_msgs::GPSRAW::ConstPtr &msg)
     {
         follow_target(
             msg->lat,
             msg->lon,
             msg->alt
         );
         std::cout << "Sending lat: " << msg->lat << " lon: "<< msg->lon << " alt: " << msg->alt << std::endl;
     }
 };
 }   // namespace extra_plugins
 }   // namespace mavros

PLUGINLIB_EXPORT_CLASS(mavros::extra_plugins::FollowTargetPlugin, mavros::plugin::PluginBase)
```

The output is here as follow:
```
Sending lat: 473988542 lon: 85456026 alt: 495857
Sending lat: 473977421 lon: 85456068 alt: 488110
Sending lat: 473977421 lon: 85456067 alt: 488102
Sending lat: 473988542 lon: 85456027 alt: 495836
```
From this, the mavlink_receiver.cpp function should be able to receive the position and add in the timestamp? I'm not sure if the other parameters such as the vel should be included here or not. However, the code still doesn't work and I get a similar issue where the drone just flies in 1 direction continuously.

Jaeyoung-Lim, 2021-06-09:
> Is it correct for me to say that there is already a function to convert the MAVLink FOLLOW_TARGET messages into uORB messages and therefore, there shouldn't be a need for me to rewrite a function to send uORB messages but instead, I should try to get the MAVLink FOLLOW_TARGET messages?

@dylantzx The documentation in https://docs.px4.io/master/en/flight_modes/follow_me.html revolves around the use case of using the mavlink message, and not having an internal px4 app publishing the uORB messaging. While publishing uORB messages is a valid approach, just keep in mind that the docs are not relevant for your use case. 

I think what you should do first, is to publish gps address controlled (not from another UAV) and see if the behavior is correct. From your comments, it is hard to distinguish issues between your implementation and issues related to PX4. 

If you have found issues related to PX4, please post an issue to the [PX4/PX4-Autopilot](https://github.com/PX4/PX4-Autopilot) repo, since this issue doesn't seem link a simulation related issue anymore.

dylantzx, 2021-06-09:
@Jaeyoung-Lim Thank you for your help so far! I think the behavior issue is an existing issue mentioned in [PX4/PX4-Autopilot](https://github.com/PX4/PX4-Autopilot/issues/17473). I am checking with them regarding this. Thank you once again!

Jaeyoung-Lim, 2021-06-14:
Closing for now! Please reopen if you have more to discuss regarding this issue

closed 2021-06-14 by Jaeyoung-Lim