PX4/PX4-SITL_gazebo-classic #768: iris_hitl cannot fly - model is incorrect

rocketengineer1982, 2021-06-05:
I'm running Gazebo HITL simulations using a PX4.  Everything works fine until I attempt to take off, at which point the vehicle begins flipping around uncontrollably.  I'm following the instructions in the PX4 user guide exactly, and I'm running
`$ gazebo Tools/sitl_gazebo/worlds/hitl_iris.world` to start the HITL simulation.

This appears to be an issue with the rotor models, their attachments, and where the rotor forces are being applied.  One of the rotors is circling the motor the it is supposed to be attached to, and I've still not been able to locate the other 3.

Here's a screenshot showing the displaced rotor.
https://imgur.com/ENlUgIa

rocketengineer1982, 2021-06-10:
I'm using Gazebo 9.0.0 running on Ubuntu 18.04.  `px4_sitl gazebo` works fine, and the Gazebo model of the Iris drone used in SITL simulation appears correct - all of the rotors are where they should be.

Jaeyoung-Lim, 2021-06-10:
Where does the `hitl_iris.world` come from?

rocketengineer1982, 2021-06-18:
I modified `iris_hitl.sdf` to fix the physical model and prevent the simulated Iris drone from immediately flipping out and crashing.  Large portions are copied directly from `iris.sdf.jinja` but a couple tweaks were necessary to get the simulated IMU and GPS working.

There may be bugs in this that I missed - I've not yet conducted exhaustive testing.  However, this version of `iris_hitl.sdf' can fly and has working GPS and IMU.

Here is the `iris_hitl.sdf` (`Tools/sitl_gazebo/models/iris_hitl.sdf`) that worked for me:

`<sdf version='1.5'>
  <model name='iris_hitl'>
    <link name='base_link'>
      <pose>0 0 0 0 0 0</pose>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>1.5</mass>
        <inertia>
          <ixx>0.029125</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.029125</iyy>
          <iyz>0</iyz>
          <izz>0.055225</izz>
        </inertia>
      </inertial>
      <collision name='base_link_inertia_collision'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <box>
            <size>0.47 0.47 0.11</size>
          </box>
        </geometry>
        <surface>
          <contact>
            <ode>
              <min_depth>0.001</min_depth>
              <max_vel>0</max_vel>
            </ode>
          </contact>
          <friction>
            <ode/>
          </friction>
        </surface>
      </collision>
      <visual name='base_link_inertia_visual'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>model://iris/meshes/iris.stl</uri>
          </mesh>
        </geometry>
        <material>
          <script>
            <name>Gazebo/DarkGrey</name>
            <uri>file://media/materials/scripts/gazebo.material</uri>
          </script>
        </material>
      </visual>
      <gravity>1</gravity>
      <velocity_decay/>
    </link>
    <link name='/imu_link'>
      <pose>0 0 0 0 0 0</pose>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>0.015</mass>
        <inertia>
          <ixx>1e-05</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1e-05</iyy>
          <iyz>0</iyz>
[... 492 lines cut ...]
          <input_index>7</input_index>
          <input_offset>0</input_offset>
          <input_scaling>0.524</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position</joint_control_type>
        </channel>
      </control_channels>
    </plugin>
    <static>0</static>
    <plugin name='rotors_gazebo_imu_plugin' filename='libgazebo_imu_plugin.so'>
      <robotNamespace/>
      <linkName>/imu_link</linkName>
      <imuTopic>/imu</imuTopic>
      <gyroscopeNoiseDensity>0.00018665</gyroscopeNoiseDensity>
      <gyroscopeRandomWalk>3.8785e-05</gyroscopeRandomWalk>
      <gyroscopeBiasCorrelationTime>1000.0</gyroscopeBiasCorrelationTime>
      <gyroscopeTurnOnBiasSigma>0.0087</gyroscopeTurnOnBiasSigma>
      <accelerometerNoiseDensity>0.00186</accelerometerNoiseDensity>
      <accelerometerRandomWalk>0.006</accelerometerRandomWalk>
      <accelerometerBiasCorrelationTime>300.0</accelerometerBiasCorrelationTime>
      <accelerometerTurnOnBiasSigma>0.196</accelerometerTurnOnBiasSigma>
    </plugin>
  </model>
</sdf>`

Jaeyoung-Lim, 2021-06-19:
@rocketengineer1982 Could you make a Pull Request so that we can in corporate the fix?

Thanks!

rocketengineer1982, 2021-07-01:
I've created a pull request.