PX4/PX4-SITL_gazebo-classic #772: How to spawn `iris_downward_depth_camera` using single_vehicle_spawn.launch ?

adbidwai, 2021-06-19:
Hello,
I want to spawn a drone using a downward-facing depth camera for altitude-based localization. 
I think the model `iris_downward_depth_camera` from `PX4-Autopilot/Tools/sitl_gazebo/models` would be the most useful for me. I am using the `multi_uav_mavros_sitl.launch` file which inherently uses `single_vehicle_spawn.launch`.  This launch file can spawn the `iris` drone but when I set the `vehicle` argument to `iris_downward_depth_camera` it results in an error saying it can't find the jinja.sdf file (which actually doesn't exist in the given path)

Help is really appreciated.

TIA

Jaeyoung-Lim, 2021-06-22:
@adbidwai When you build the package, the sdf files will be generated using the jinja templates

adbidwai, 2021-06-23:
@Jaeyoung-Lim @LorenzMeier  What are the jinja templates and how does it generate the SDF files? Also, where are they stored?
I still did not understand how exactly am I supposed to use the models in ```PX4-Autopilot/Tools/sitl_gazebo/models```

Jaeyoung-Lim, 2021-07-03:
@adbidwai You can look up on what jinja templates are in: https://jinja.palletsprojects.com/en/3.0.x/

The sdf files are generated through jinja templates at build time, and then the generated sdf files can be used as normal sdf files.

closed 2021-07-19 by Jaeyoung-Lim