PX4/PX4-SITL_gazebo-classic #774: Gazebo's world not spawning

ZhiWei20011, 2021-06-23:
Hi, I tried to download your mcmillan_airfield world file to my workspace and my gazebo is not spawning (it keeps saying waiting for simulation in the terminal) when I called my workspace to launch the mcmillan world. Any suggestions? Thank you.

Jaeyoung-Lim, 2021-06-23:
@ZhiWei20011 Please provide more details on what you are trying to do.

Could you confirm everything works with the standard setup?
(e.g. `make px4_sitl gazebo_iris__mcmillain_airfield`)

ZhiWei20011, 2021-06-25:
Nope, I am not able to launch it with the standard setup. Any reason why?
[image]

Jaeyoung-Lim, 2021-06-25:
@ZhiWei20011 I don't think what you are running is a supported command.

Please follow the documentation to try out this repoistory: http://docs.px4.io/master/en/simulation/gazebo.html

closed 2021-07-19 by Jaeyoung-Lim