PX4/PX4-SITL_gazebo-classic #777: Error at launch  "make px4_sitl gazebo"

Y-Grigoriy, 2021-06-30:
Hello,

I try to use Gazebo simulation following instructions on the website: http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets and http://docs.px4.io/master/en/dev_setup/building_px4.html However I got following errors:
```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/grigoriy/Documents/PX4-Autopilot/build/px4_...me/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/grigoriy/Documents/PX4-Autopilot
build_path: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :$/Documents/drone-games/gazebo/build:/home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :$/Documents/drone-games/gazebo/modelssource:/home/grigoriy/Documents/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/grigoriy/Documents/new_ws/devel/lib:/opt/ros/noetic/lib:/home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /home/grigoriy/Documents/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Gazebo multi-robot simulator, version 11.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 127.0.0.1
[Msg] Loading world file [/home/grigoriy/Documents/PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world]
[Wrn] [gazebo_gps_plugin.cpp:78] [gazebo_gps_plugin]: iris::gps0 using gps topic "gps0"
[Wrn] [gazebo_gps_plugin.cpp:203] [gazebo_gps_plugin] Using default update rate of 5hz
[Err] [Plugin.hh:178] Failed to load plugin /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: undefined symbol: _ZTIN8mav_msgs4msgs17CommandMotorSpeedE
[Err] [Plugin.hh:178] Failed to load plugin /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: undefined symbol: _ZTIN8mav_msgs4msgs17CommandMotorSpeedE
[Err] [Plugin.hh:178] Failed to load plugin /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: undefined symbol: _ZTIN8mav_msgs4msgs17CommandMotorSpeedE
[Err] [Plugin.hh:178] Failed to load plugin /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_motor_model.so: undefined symbol: _ZTIN8mav_msgs4msgs17CommandMotorSpeedE
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
gzserver: symbol lookup error: /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so: undefined symbol: _ZN8mav_msgs4msgs17CommandMotorSpeedC1Ev
SITL COMMAND: "/home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/grigoriy/Documents/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/etc -> /home/grigoriy/Documents/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 406 bytes, decoded 406 bytes
[param] Loaded: eeprom/parameters_10016
Gazebo multi-robot simulator, version 11.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
^CERROR [px4] Startup script returned with return value: 2
/home/grigoriy/Documents/PX4-Autopilot/Tools/sitl_run.sh: line 244: kill: (17925) - No such process
ninja: build stopped: interrupted by user.
make: *** [Makefile:233: px4_sitl] Interrupt
```
If I right understand, I have problems with gazebo plugins and connection to gazebo. My system:
Windows 10, WSL2 with Ubuntu 20.04.2 LTS and using VcWsrv
Gazebo multi-robot simulator, version 11.6.0
Also I have ROS noetic 1.15.11
VcWsrv was adjusted in accordance with the following video: https://www.youtube.com/watch?v=DW7l9LHdK5c

Could you help me?

Jaeyoung-Lim, 2021-07-05:
@Y-Grigoriy I think your problem is related with your WSL environment. Have you followed the instructions onhttps://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html for your WSL development environment?

Y-Grigoriy, 2021-07-05:
That is, the problem cannot be solved at the moment? I have not tried instruction on the link.

Jaeyoung-Lim, 2021-07-05:
@Y-Grigoriy Your problem is with your environment. Please set the environment up as the documentation.

Y-Grigoriy, 2021-07-06:
@Jaeyoung-Lim, your solution does not suit me, because it does not include usage of ROS (at least in instruction). I would prefer to use Ubuntu in WSL with ros-gazebo package, however if I right understand it does not work for my current installation configuration. May it work if I will use WSL2  + Ubuntu 18 + gazebo + ros melodic (instead of noetic)?

Jaeyoung-Lim, 2021-07-07:
> May it work if I will use WSL2 + Ubuntu 18 + gazebo + ros melodic (instead of noetic)?

@Y-Grigoriy I am not sure, since ROS / PX4 with WSL is currently not supported. 

If you could get it to work and contribute to the px4 documentation, this would be helpful for other people trying to use the same environment.

closed 2021-07-28 by Jaeyoung-Lim