PX4/PX4-SITL_gazebo-classic #779: Error at launching iris_downward_depth_camera vehicle

adbidwai, 2021-07-07:
## System configurations
Ubuntu 18.04 with ROS Melodic, latest PX4-Autopilot build

## Issue
When I try launching ```iris_dwonward_depth_camera``` using ```roslaunch px4 single_vehicle_spawn_sdf.launch vehicle:=iris_downward_depth_camera``` , I get the following error.

```
roslaunch px4 single_vehicle_spawn_sdf.launch vehicle:=iris_downward_depth_camera
... logging to /home/adbidwai/.ros/log/7cad640a-debe-11eb-9c87-2016b98536e4/roslaunch-adbidwai-g3-3579-29293.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://adbidwai-g3-3579:42495/

SUMMARY
========

PARAMETERS
 * /model_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    iris_downward_depth_camera_1_spawn (gazebo_ros/spawn_model)
    sitl_1 (px4/px4)

ROS_MASTER_URI=http://localhost:11311

process[sitl_1-1]: started with pid [29309]
INFO  [px4] Creating symlink /home/adbidwai/sop_ws/src/PX4-Autopilot/build/px4_sitl_default/etc -> /home/adbidwai/.ros/sitl_iris_downward_depth_camera_1/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 1
Error: Unknown model iris_downward_depth_camera (not found by name on /home/adbidwai/.ros/sitl_iris_downward_depth_camera_1/etc/init.d-posix/airframes)
ERROR [px4] Startup script returned with return value: 256
process[iris_downward_depth_camera_1_spawn-2]: started with pid [29325]
[sitl_1-1] process has died [pid 29309, exit code 255, cmd /home/adbidwai/sop_ws/src/PX4-Autopilot/build/px4_sitl_default/bin/px4 /home/adbidwai/sop_ws/src/PX4-Autopilot/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -i 1 -w sitl_iris_downward_depth_camera_1 -d __name:=sitl_1 __log:=/home/adbidwai/.ros/log/7cad640a-debe-11eb-9c87-2016b98536e4/sitl_1-1.log].
log file: /home/adbidwai/.ros/log/7cad640a-debe-11eb-9c87-2016b98536e4/sitl_1-1*.log
[INFO] [1625622030.002357, 0.000000]: Loading model XML from ros parameter model_description
[INFO] [1625622030.008028, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
```
[image]

### Help will be appreciated, TIA

adbidwai, 2021-07-07:
@Jaeyoung-Lim Sir, request to please help out

Jaeyoung-Lim, 2021-07-07:
Try 
```
roslaunch px4 single_vehicle_spawn_sdf.launch vehicle:=iris_downward_depth_camera PX4_SIM_MODEL:=iris
```

adbidwai, 2021-07-07:
@Jaeyoung-Lim same issue
[image]

Jaeyoung-Lim, 2021-07-07:
@adbidwai Can you manually set the `PX4_SIM_MODEL=iris` parameter in by editing your launchfile?

Jaeyoung-Lim, 2021-07-07:
@adbidwai Does this work at all for the default models on your system?

adbidwai, 2021-07-07:
@Jaeyoung-Lim Actually I'll try to tell you the final aim, I want to run a multi UAV simulation with every drone with a camera publishing its camera feed on a ROS topic? Any suggestions on how could I achieve that?

Jaeyoung-Lim, 2021-07-07:
@adbidwai Could you answer my question first, so we can solve this issue

adbidwai, 2021-07-07:
Sorry @Jaeyoung-Lim 
I think I did not understand your question, 
When I just run ```roslaunch px4 single_vehicle_spawn_sdf.launch``` without any changes in the launch file, I get the following in my terminal without Gazebo opening
```
roslaunch px4 single_vehicle_spawn_sdf.launch
... logging to /home/adbidwai/.ros/log/b416a178-df21-11eb-a844-2016b98536e4/roslaunch-adbidwai-g3-3579-14527.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://adbidwai-g3-3579:45605/

SUMMARY
========

PARAMETERS
 * /model_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    plane_1_spawn (gazebo_ros/spawn_model)
    sitl_1 (px4/px4)

auto-starting new master
process[master]: started with pid [14559]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b416a178-df21-11eb-a844-2016b98536e4
process[rosout-1]: started with pid [14571]
started core service [/rosout]
process[sitl_1-2]: started with pid [14577]
process[plane_1_spawn-3]: started with pid [14579]
INFO  [px4] Creating symlink /home/adbidwai/sop_ws/src/PX4-Autopilot/build/px4_sitl_default/etc -> /home/adbidwai/.ros/sitl_plane_1/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 1
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  MAV_SYS_ID: curr: 1 -> new: 2
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4561
[INFO] [1625662194.154934]: Loading model XML from ros parameter model_description
[INFO] [1625662194.181123]: Waiting for service /gazebo/spawn_sdf_model
```

closed 2021-07-15 by adbidwai

Jaeyoung-Lim, 2021-07-15:
@adbidwai How did you solve your problem?

adbidwai, 2021-07-15:
Hi, @Jaeyoung-Lim 
I could not launch the vehicle `iris_downward_depth_camera` using the `single_vehicle_spawn_sdf.launch` till the end. :(
The turnaround I used to use the `iris_downward_depth_camera` model was: I copied the included `depth_camera`'s SDF in my `iris.sdf.jinja` and initialized appropriate joints so it pointed downward (basically I took the reference of `iris_downward_depth_camera.sdf`'s reference). (I could have also created a jinja template separately for `iris_downward_depth_camera` but I did not know how to make one :sweat_smile: , hence I did it this way)
I then used the file `multi_uav_mavros_sitl.launch` / `single_vehicle_spawn.launch` to launch my iris drone with a downward pointing depth camera.

adbidwai, 2021-07-15:
@Jaeyoung-Lim Thanks for your help though!

kaushal07wick, 2023-02-27:
> Hi, @Jaeyoung-Lim I could not launch the vehicle `iris_downward_depth_camera` using the `single_vehicle_spawn_sdf.launch` till the end. :( The turnaround I used to use the [...]

hey @adbidwai can u explain this a bit more, i am aslo trying to launch two drones with front facing depth camera.
prompt assistance would be great.