PX4/PX4-SITL_gazebo-classic #780: Support ignition gazebo

blakermchale, 2021-07-11:
Supporting ignition gazebo would be useful since it's looking like classic gazebo will be replaced inevitably. There's already a migration tutorial for ardupilot [here](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/tutorials/migrating_ardupilot_plugin.md). I'm not sure how difficult this would be or if it would be useful. I do know ignition gazebo has support for separated cloud based simulations and would definitely expand the usability of PX4 across the community.

LorenzMeier, 2021-07-11:
We have in fact completed a first iteration. @Jaeyoung-Lim what's the current state to make this available to the wider community?

Jaeyoung-Lim, 2021-07-11:
@LorenzMeier I am coordinating with @machtelik in making the repo public!

Jaeyoung-Lim, 2021-07-12:
@blakermchale A little sneakpeak of the current status of the ignition implementation: https://youtu.be/38UJqrNQChg

Jaeyoung-Lim, 2021-07-12:
Pull request created in https://github.com/PX4/PX4-Autopilot/pull/17889

Therefore, closing this issue

closed 2021-07-12 by Jaeyoung-Lim