PX4/PX4-SITL_gazebo-classic #783: Mavlink connection does not start for custom quadrotor

Y-Grigoriy, 2021-07-14:
I prepared new model of quadrotor in Gazebo following to the next steps:
1. Create _my_vehicle_ model of the quadrotor based on the files of iris model (.config and .sdf files with folder meshes).
2. _my_vehicle_ name was added to: PX4-Autopilot/platforms/posix/cmake/sitl_target.cmake
   set(models none iris iris_opt_flow ... rover hippocampus <newmodel name>)
3. Create an airframe file under ROMFS/px4fmu_common/init.d-posix (based off the iris files), give it a number (10019) and name it _10019_my_vehicle_
4. I added _my_vehicle_ in CMakeLists.txt (PX4-Autopilot/ROMFS/px4fmu_common/init.d-posix/airframes/)

References:
https://github.com/PX4/PX4-Devguide/issues/398
https://discuss.px4.io/t/create-custom-model-for-sitl/6700/16

However, gazebo server does not launch automatically after command: `make px4_sitl gazebo_my_vehicle`
```
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:135] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo
```
I started gzserver in additional terminal and model was shown in Gazebo environment, however the Mavlink connection was not established:
```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/user/PX4-Autopilot/build...er/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/user/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: my_vehicle
world: none
src_path: /home/user/PX4-Autopilot
build_path: /home/user/PX4-Autopilot/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/user/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/user/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/user/uav-flight-simulation/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/user/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /home/user/PX4-Autopilot/Tools/sitl_gazebo/models/my_vehicle/my_vehicle.sdf
SITL COMMAND: "/home/user/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/user/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/user/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/user/PX4-Autopilot/build/px4_sitl_default/etc -> /home/user/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10019
INFO  [param] selected parameter default file eeprom/parameters_10019
[param] parameter file not found, creating eeprom/parameters_10019
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10019
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
Gazebo multi-robot simulator, version 9.18.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.18.0.3
[Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-my_vehicle'
```
Could you help me?

Jaeyoung-Lim, 2021-07-15:
@Y-Grigoriy Did you modify the copied over `iris` model so that the referenced meshes are also looking for the meshes inside `myvehicle`?

Y-Grigoriy, 2021-07-15:
@Jaeyoung-Lim `my_vehicle` is displayed correctly in Gazebo (links to new meshes have been changed correctly, both for the robot body and propellers, and also I updated the rotor joints), the problem is that the autopilot simulation does not start.

Jaeyoung-Lim, 2021-07-15:
> the problem is that the gazebo server does not launch automatically and autopilot simulation does not start.

@Y-Grigoriy In fact it does: you can see that gzserver has started successfully
```
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.18.0.3
```

What you can do is try running the following command for gazebo be more explicit about the failures
```
VERBOSE_SIM=1 make px4_sitl gazebo_myvehicle
```

Y-Grigoriy, 2021-07-16:
@Jaeyoung-Lim, in my initial message I mentioned, that gazebo server did not start and I launched it in additional terminal. Nevertheless, you helped me. I checked launching with _VERBOSE_SIM=1_ and found that `gazebo_multirotor_base_plugin` had wrong **link on the robot body**. I fixed it, and now my_vehicle can fly)
Thank you.

closed 2021-07-16 by Y-Grigoriy