PX4/PX4-SITL_gazebo-classic #786: Parachute deployment does not appear to work

hamishwillee, 2021-07-22:
The instructions in https://docs.px4.io/master/en/simulation/gazebo.html#simulated-parachute-flight-termination explain that you can simulate parachute deployment by first launching the simulator with:
```
make px4_sitl gazebo_if750a
```
Then force flight termination.

I tested this by forcing an altitude safety geofence violation (set a low altitude ceiling and flew straight up). 
The flight termination happens - there are GCS notifications and the vehicle crashes. But there is no indication that a parachute is deployed. 

How would you know? Shouldn't we see something appear on the screen?

Jaeyoung-Lim, 2021-07-23:
@hamishwillee Thanks for posting this issue,

The main reason is that the pwm values of the flight termination has changed on the firmware side, and we need to adjust the parachute threshold value.

However, when trying to fix the problem I noticed also that the firmware is emitting termination pwm values when arming. 
Therefore the parachute is being deployed before takeoff. 

Since there is no way from the simulation side to distinguish flight termination and a normal actuator output now, I will try to talk to somebody in the flight control team to see if we can fix the problem

bresch, 2021-07-26:
@Jaeyoung-Lim Something might have changed on the Firmware side but I'm not aware of it. At boot, the PWM signals should go to their "disarmed" and not "failsafe" values.

Jaeyoung-Lim, 2021-07-27:
@bresch I think this wasn't the case before, but now the disarmed value and failsafe values have become identical. Would you have any idea what might have introduced this change?

bresch, 2021-07-28:
> Would you have any idea what might have introduced this change?

Nope... :(

akshata-1, 2021-09-29:
@Jaeyoung-Lim Is this issue resolved?

Jaeyoung-Lim, 2021-09-29:
@akshata-1 No, unfortunately I was not able to fix it

iamrajee, 2022-04-20:
I'm facing the same issue. Is there any update? @Jaeyoung-Lim