PX4/PX4-SITL_gazebo-classic #788: Yosemite world fails to launch in gazebo

mihirk2309, 2021-07-27:
I am trying to launch gazebo with the Yosemite world. I am using 
`make px4_sitl_default gazebo___yosemite `
command
But Gazebo is stuck at this point and does not open even after waiting for a really long time.
[image]
For all other worlds like sonoma_raceway or baylands etc. gazebo opens up flawlessly without having to wait for even a few seconds. Can anyone help with this issue?

tfoote, 2021-07-27:
If you're only waiting for less than a few seconds for the others you are likely using a cached copy that's local. When the splash screen is loading like that it's fetching the assets for the world. If you see network traffic you may just need to be more patient. The Yosemite world is a lot larger, and may take some time to download depending on your bandwidth availability.

mihirk2309, 2021-07-28:
I don't think that's the issue. Here's why:
1) I waited for more than 2 hours for it to load but it did not.
2) Network bandwidth is sufficient (200mbps stable)
3) I tried the same after doing `rosclean purge` and cloning a fresh PX4_Autopilot but its still stuck at this point
[image]

Jaeyoung-Lim, 2021-07-28:
@mihirk2309 You can try removing your local cache and enable `VERBOSE_SIM=1` to see more information of what the problem is

mihirk2309, 2021-07-28:
Deleting all Yosemite related files from `~/.gazebo` and building again did the job. Thanks a lot, @Jaeyoung-Lim and @tfoote for your help!

closed 2021-07-28 by mihirk2309