PX4/PX4-SITL_gazebo-classic #789: PX4 Gazebo Unable to render textures.

TomoCoder123, 2021-08-04:
I am running Gazebo alongside PX4 and in the custom world I created, it seems that the textures are not loading in for the trees (pine tree model). It seems to use images from my screen to create the textures as sometimes Gazebo Menus can be shown on the tree textures in place of the white areas. The white poles on the left are actually trees and they should look more like the trees on the right. Also the environment in the back should just be the VRC driving course from the gazebo models but that is also not rendered.
[image]
Thanks in Advance

Jaeyoung-Lim, 2021-08-04:
@TomoCoder123 Clearly gazebo is rendering textures on the ground and the leaves on the trees. What are the differences?

TomoCoder123, 2021-08-04:
The trees on the left are not being rendered at all (the white poles are supposed to be trees). The following image represents the problem more clearly.
[image]
If you look closely you can see characters on the tree trunk which it seems to just be an image of the terminal. This only seems to be a problem when running PX4 as when I am using gazebo it does not occur

Jaeyoung-Lim, 2021-08-04:
@TomoCoder123 The green leaves that you see and the grass on the floor ARE textures. Therefore the rendering is working, I suspect that it is unable to find the texture that it is referencing.

Which environment is this? Is this one of the worlds that are part of this repository?

TomoCoder123, 2021-08-04:
The VRC driving course and the pine trees are both from gazebos default models and this is a custom world. However the custom world seems to work on other’s computers as two partners of mine both had the models working fine.

tfoote, 2021-08-17:
That sounds like you do not have the resources in the correct path for the Gazebo instance launched by the PX4 stack to be able to load the models appropriately on your computer. You may have injected the trees into various paths manually which are not persisted by the way you're launching Gazebo via PX4.

Jaeyoung-Lim, 2021-09-21:
@TomoCoder123 Please reopen if you think this is still an issue. Closing as stale

closed 2021-09-21 by Jaeyoung-Lim