PX4/PX4-SITL_gazebo-classic #792: Iris lidar not showing anything in gazebo_sitl_multiple_run

saadnec, 2021-08-16:
i made a sdf model file to connect lidar with my iris drone. The code is given below.
"""
<?xml version="1.0" ?>
<sdf version='1.5'>
  <model name='iris_lidar'>

    <include>
      <uri>model://iris</uri>
    </include> 

    <include>
      <uri>model://lidar</uri>
      <pose>-0.12 0 0 0 3.1415 0</pose>
    </include>
    <joint name="lidar_joint" type="fixed">
      <child>lidar::link</child>
      <parent>iris::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

    <include>
      <uri>model://flow_cam</uri>
      <pose>-0.12 0.05 0 0 1.5708 1.5708</pose>
    </include>
    <joint name="flow_joint" type="revolute">
      <child>camera::link</child>
      <parent>iris::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

    <include>
      <uri>model://lidar</uri>
      <pose>0 0 0.1 0 0 0</pose>
    </include>
    <joint name="lidar_joint" type="revolute">
      <child>lidar::link</child>
      <parent>iris::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

  </model>
</sdf>

""""
Changed vehicle model from iris to iris_lidar
vehicle_model=${VEHICLE_MODEL:="iris_lidar".

But itsn't showing anything when i launch the gazebo_sitl_multiple_run file no vehicle is being shown. What am i doing wrong?

Jaeyoung-Lim, 2021-08-16:
@saadnec Models with `<include>` tags do not support multiple vehicle scripts. You would need directly include the lidar sensor into the [jinja template](https://github.com/PX4/PX4-SITL_gazebo/blob/master/models/iris/iris.sdf.jinja) in order to run multiple vehicles with the sensor:

saadnec, 2021-08-16:
Thanks

> 
> 
> @saadnec Models with `<include>` tags do not support multiple vehicle scripts. You would need directly include the lidar sensor into the [jinja template](https://github.com/PX4/PX4-SITL_gazebo/blob/master/models/iris/iris.sdf.jinja) in order to run multiple vehicles with the sensor:

Thanks for the response, now they are showing after making that change but now i am getting multiple warnings which  i wasnt facing earlier before connecting the lidar with it.

""""
Err] [Model.cc:272] LoadJoint Failed
[Err] [Joint.cc:263] EXCEPTION: Couldn't Find Parent Link[iris::base_link]

[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call

[Wrn] [gazebo_lidar_plugin.cpp:126] [gazebo_lidar_plugin]: iris_lidar_0::lidar using lidar topic "lidar"
[Wrn] [RaySensor.cc:289] ranges not constructed yet (zero sized)

""""
Now i am not even able to connect with the vehicles using UDP. Whereas gazebo_sitl_multiple_run's code is the same just changed the vehicle's setting

Jaeyoung-Lim, 2021-08-16:
@saadnec You probably have something wrong with your sdf file
```
[Err] [Joint.cc:263] EXCEPTION: Couldn't Find Parent Link[iris::base_link]
```
Therefore the model is not even being spawned

saadnec, 2021-08-16:
> 
> 
> @saadnec You probably have something wrong with your sdf file
> 
> ```
> [Err] [Joint.cc:263] EXCEPTION: Couldn't Find Parent Link[iris::base_link]
> ```
> 
> Therefore the model is not even being spawned
""""""
<?xml version="1.0" ?>
<sdf version='1.5'>
  <model name='iris_lidar'>

    <include>
      <uri>model://iris</uri>
    </include> 

    <include>
      <uri>model://lidar</uri>
      <pose>-0.12 0 0 0 3.1415 0</pose>
    </include>
    <joint name="lidar_joint" type="fixed">
      <child>lidar::link</child>
      <parent>iris_lidar::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

    <include>
      <uri>model://flow_cam</uri>
      <pose>-0.12 0.05 0 0 1.5708 1.5708</pose>
    </include>
    <joint name="flow_joint" type="revolute">
      <child>camera::link</child>
      <parent>iris_lidar::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

    <include>
      <uri>model://lidar</uri>
      <pose>0 0 0.1 0 0 0</pose>
    </include>
    <joint name="lidar_joint" type="revolute">
      <child>lidar::link</child>
      <parent>iris_lidar::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

  </model>
</sdf>

""""

This is my sdf file i changed  <parent>iris::base_link</parent> to  <parent>iris_lidar::base_link</parent> but still the errors are still the same.

Jaeyoung-Lim, 2021-08-16:
@saadnec Are you sure the models you are generaing through the `gazebo_sitl_multiple_run` are called `iris`? and not `iris_0` ?

Jaeyoung-Lim, 2021-08-16:
@saadnec As I said, you need to make the changes directly in https://github.com/PX4/PX4-SITL_gazebo/blob/master/models/iris/iris.sdf.jinja

saadnec, 2021-08-16:
Yes i followed your instructions and made the required changes too. After making the changes my iris.sdf.ninja file looks like this

""""

<sdf version='1.6'>
  <model name='iris'>
    <link name='base_link'>
      <pose>0 0 0 0 0 0</pose>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>1.5</mass>
        <inertia>
          <ixx>0.029125</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.029125</iyy>
          <iyz>0</iyz>
          <izz>0.055225</izz>
        </inertia>
      </inertial>
      <collision name='base_link_inertia_collision'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <box>
            <size>0.47 0.47 0.11</size>
          </box>
        </geometry>
        <surface>
          <contact>
            <ode>
              <min_depth>0.001</min_depth>
              <max_vel>0</max_vel>
            </ode>
          </contact>
          <friction>
            <ode/>
          </friction>
        </surface>
      </collision>
      <visual name='base_link_inertia_visual'>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <scale>1 1 1</scale>
            <uri>model://iris/meshes/iris.stl</uri>
          </mesh>
        </geometry>
        <material>
          <script>
            <name>Gazebo/DarkGrey</name>
            <uri>file://media/materials/scripts/gazebo.material</uri>
          </script>
        </material>
      </visual>
      <gravity>1</gravity>
      <velocity_decay/>
    </link>
    <link name='/imu_link'>
      <pose>0 0 0 0 0 0</pose>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>0.015</mass>
        <inertia>
          <ixx>1e-05</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1e-05</iyy>
          <iyz>0</iyz>
          <izz>1e-05</izz>
        </inertia>
[... 537 lines cut ...]
          <input_offset>0</input_offset>
          <input_scaling>0.524</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position</joint_control_type>
        </channel>
      </control_channels>
    </plugin>
    <static>0</static>
    <plugin name='rotors_gazebo_imu_plugin' filename='libgazebo_imu_plugin.so'>
      <robotNamespace/>
      <linkName>/imu_link</linkName>
      <imuTopic>/imu</imuTopic>
      <gyroscopeNoiseDensity>0.00018665</gyroscopeNoiseDensity>
      <gyroscopeRandomWalk>3.8785e-05</gyroscopeRandomWalk>
      <gyroscopeBiasCorrelationTime>1000.0</gyroscopeBiasCorrelationTime>
      <gyroscopeTurnOnBiasSigma>0.0087</gyroscopeTurnOnBiasSigma>
      <accelerometerNoiseDensity>0.00186</accelerometerNoiseDensity>
      <accelerometerRandomWalk>0.006</accelerometerRandomWalk>
      <accelerometerBiasCorrelationTime>300.0</accelerometerBiasCorrelationTime>
      <accelerometerTurnOnBiasSigma>0.196</accelerometerTurnOnBiasSigma>
    </plugin>
  </model>
</sdf>
""""

saadnec, 2021-08-17:
I have finally implemented it and the drones are showing in my gazebo_sitl_multiple_run world but i am having these errors now  due to which i am unable to connect to my vehicles using udp.
Errors while connecting to vehicle are
"""
[10:40:03|Error] No known remotes (udp_connection.cpp:133)
[10:40:03|Error] send fail (mavsdk_impl.cpp:160)
"""
And the ones after launching my world are
""""
Spawning iris_2 at 0.0 6
Warning [parser.cc:723] Can not find the XML attribute 'version' in sdf XML tag for model: rplidar. Please specify the SDF protocol supported in the model configuration file. The first sdf tag in the config file will be used 
Starting gazebo client
Warning [parser.cc:723] Can not find the XML attribute 'version' in sdf XML tag for model: rplidar. Please specify the SDF protocol supported in the model configuration file. The first sdf tag in the config file will be used 
[Wrn] [gazebo_gps_plugin.cpp:78] [gazebo_gps_plugin]: iris_2::gps0 using gps topic "gps0"
[Wrn] [gazebo_gps_plugin.cpp:203] [gazebo_gps_plugin] Using default update rate of 5hz 
[Err] [Plugin.hh:178] Failed to load plugin libGpuRayPlugin.so: libGpuRayPlugin.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:178] Failed to load plugin libgazebo_ros_gpu_laser.so: libgazebo_ros_gpu_laser.so: cannot open shared object file: No such file or directory
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
[Wrn] [gazebo_mavlink_interface.cpp:152] No identifier on joint. Using 0 as default sensor ID

""""

closed 2021-08-22 by saadnec