PX4/PX4-SITL_gazebo-classic #793: roslaunch px4 single_vehicle_spawn_sdf.launch not working

saadnec, 2021-08-20:
I am running the following command "roslaunch px4 single_vehicle_spawn_sdf.launch not working" but it produces the error
''''
RLException:[single_vehicle_spawn_sdf.launch] is neither a launch file in package [px4] nor is a[px4] a launch file name
The traceback for excpetion was written to the log file
"""""

Jaeyoung-Lim, 2021-08-21:
@saadnec Your failure has nothing to do with the launchfile itself, but because you have not configured your ros package path properly (since it cannot find px4)

saadnec, 2021-08-22:
> 
> 
> @saadnec Your failure has nothing to do with the launchfile itself, but because you have not configured your ros package path properly (since it cannot find px4)

Thanks, i configured the ros package path but now its showing the following error
""""
RLException: Invalid <param> tag: Cannot load command parameter [model_description]: no such command [['xmlstarlet', 'ed', '-d', '//plugin[@name="mavlink_interface"]/mavlink_tcp_port', '-s', '//plugin[@name="mavlink_interface"]', '-t', 'elem', '-n', 'mavlink_tcp_port', '-v', '4560', '/home/saad/Firmware/Tools/sitl_gazebo/models/plane/plane.sdf']]. 

Param xml is <param command="$(arg cmd)" name="model_description"/>
"""
Does it have something to do with not configuring properly too?

closed 2021-08-22 by saadnec