PX4/PX4-SITL_gazebo-classic #798: Gazebo's simulation compile errors on MAC

kevinuav, 2021-09-05:
I encounter gazebo errors while it is compiling with my Mac.I ran gazebo on my CatalinaOS v10.5.1 very well before. But it can naver compile successfully after that I updated my code from v1.11.0 to v1.12.0. Even I have formatted the whole disk of my Mac.Then I reinstalled everything as the instruction carefully. It still get the very same errors as before.

Jaeyoung-Lim, 2021-09-05:
> I encounter gazebo errors while it is compiling with my Mac.I ran gazebo on my CatalinaOS v10.5.1 very well before. But it can naver compile successfully after that I updated my code from v1.11.0 to v1.12.0.

@kevinuav Does this mean that you can compile successfully with `v1.11.0` on your current setup?

kevinuav, 2021-09-06:
> > I encounter gazebo errors while it is compiling with my Mac.I ran gazebo on my CatalinaOS v10.5.1 very well before. But it can naver compile successfully after that I updated my code from v1.11.0 to v1.12.0.
> 
> @kevinuav Does this mean that you can compile successfully with `v1.11.0` on your current setup?
> 
> Related to #775 #769

No,I switch back to v1.11.3. It still can't compile successfully. But the errors are not the same with v1.12.0
[v1.11.3 gazebo's compiling output.odt](https://github.com/PX4/PX4-SITL_gazebo/files/7114534/v1.11.3.gazebo.s.compiling.output.odt)

kevinuav, 2021-09-14:
I have fixed the problem after many Time cleaned up and installed everything on my MAC. The problem is caused by the Mac's self build-in pythons. After it has been cleaned up. Then the gazebo works!
here's the related issue: https://gist.github.com/MuhsinFatih/ee0154199803babb449b5bb98d3475f7

closed 2021-09-14 by kevinuav