PX4/PX4-SITL_gazebo-classic #799: Sitl standard VTOL problem V1.12

moreba1, 2021-09-08:
Hi

I can not do SITL standard VTOL with version of 1.12.0 to 1.12.3 but I can it with version of 1.11.3 .
gazebo and SITL runs but it does not connect to QGroundControl. I don't have this problem with simulation of plane and multirotor.

Jaeyoung-Lim, 2021-09-19:
@moreba1 
- I can run the standard VTOL with the latest version of PX4, 
- Standard vtol is being checked as part of the SITL testing automatically in the CI, therefore this would have been detected in the firmware CI but it hasn't.

Therefore, Can you recheck your environment?

Jaeyoung-Lim, 2021-09-21:
@moreba1 Please reopen if this is still an issue. Closing since I didn't get a response.

closed 2021-09-21 by Jaeyoung-Lim