PX4/PX4-SITL_gazebo-classic #800: CMakeLists.txt not configured to install physics_msgs

langthorsten, 2021-09-09:
I just want to inform you that the CMakeLists.txt file needs to be updated to include physics_msgs:
`install(TARGETS ${plugins} mav_msgs nav_msgs std_msgs sensor_msgs physics_msgs DESTINATION ${PLUGIN_PATH})`

This is important if one wants to install SITL_gazebo system wide, independently of PX4-Autopilot.

Jaeyoung-Lim, 2021-09-09:
@langthorsten Could you explain why would one want to install sitl_gazebo system wide? What would be the use case?

langthorsten, 2021-09-10:
@Jaeyoung-Lim It's easier to use. One doesn't need to set so many environment variables and multiple users can access it. Most gazebo plugin packages can be installed system wide. My personal use case is running gazebo in a docker container.

I'm actually not asking for a new feature. I just wanted to inform you that the current CMakeLists.txt ist broken, because `physics_msgs` was missing in the install line. If you don't want to support the CMake install target, you should probably remove the line completely, because otherwise the installed package will produce errors.

ankyAgrawal, 2024-06-13:
I am facing the same issue. Physics_msgs missing. How do I install it by myself?