PX4/PX4-SITL_gazebo-classic #806: setpoint_raw/attitude issue, the fixed wind does not follow heading input

joaoalves10, 2021-09-21:
Hi, i am trying to give a heading command to my fixed wing in ROS/Gazebo. When i use the following message:
rostopic pub -r 5 /mavros/setpoint_raw/attitude mavros_msgs/AttitudeTarget "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
type_mask: 0b10000000
orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
body_rate: {x: 0.1, y: 0.0, z: -0.1}
thrust: 0.5"

the airplane starts to turn but does not execute a loiter, it flys like the image down bellow.
[image]

Also, when i put the body_rate z = 0,1 it starts to turn right, and when i change it to z=-0,1 it also turns right.
I do not understand why,
i put the type_mask 0b10000000 to ignore orientation.

If someone could help me i would appreciate it

cheers

Jaeyoung-Lim, 2021-09-21:
@joaoalves10 Thank you for reporting the issue, this repository is for the gazebo simulation. However, your issue doesn't seem to be simulation related but more of a question regarding how to use offboard mode using mavros. A more appropriate place to post this question would be https://discuss.px4.io/

If you are unsure on how to use mavros, you can look up the mavros wiki: http://wiki.ros.org/mavros

Nevertheless, I can see a few problems with your setpoint so I will comment.

`type_mask: 0b10000000` : you are configuring the setpoint to ignore attitude (see http://docs.ros.org/en/api/mavros_msgs/html/msg/AttitudeTarget.html ) Therefore you are not setting the setpoint for heading control at all.

`orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}` :You need to set the attitude of the vehicle to set the heading. However, not that this is a full attitude setpoint, not just the heading. It is not possible to just set the heading.

`body_rate: {x: 0.1, y: 0.0, z: -0.1}` : body rate means body angular rates, not the heading. So there seems to be a mismatch between what you are saying and what you are sending. So it will basically roll and yaw at 0.1 rad/s which is likely to not end up in a loiter.

joaoalves10, 2021-09-21:
Thank you so much for your comment. it was really helpful.

I will post it in the link that you sent me.

Nevertheless, i would like, if possible, to clarify some of the points that you talked about.

-the quaternion are 3D coordinates, could you give me a specific example to control the attitude?

- my objective to this project is being able to send commandas of heading and the plane follow the new heading. For that i would be changing the body_rate z periodicly.

Jaeyoung-Lim, 2021-09-22:
> -the quaternion are 3D coordinates, could you give me a specific example to control the attitude?

They are not 3D, : https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation

> my objective to this project is being able to send commandas of heading and the plane follow the new heading. For that i would be changing the body_rate z periodicly.

Changing body_rate z means rotating around the yaw axis. Therefore if you are in a roll, you will change both pitch and yaw if you change `body_rate_z`

joaoalves10, 2021-09-22:
@Jaeyoung-Lim thats my objective, i want to change the z and make the aircraft changing both heading and roll. In this case, the way that i was writting the message was the right one?

Jaeyoung-Lim, 2021-09-22:
@joaoalves10 No, it will not loiter since you are pitching up also.

joaoalves10, 2021-09-22:
@Jaeyoung-Lim So if my goal is to have a fixed wing flying straight and then i want to send a command for it to turn, for example, 30º degrees right, in other words, changing from 360º to 30º heading, what is the mavros message that i should use?
i thought it should be setpoint_raw/attitude, but i do not have that much knowledge about the topic. Do you know the command that i should use?

Jaeyoung-Lim, 2021-09-22:
@joaoalves10 As I mentioned above, you dont have a command available.

You can either send attitude setpoints and handle the flight controls yourself, or send position setpoints far away so that the vehicle flies on the direction you want.

joaoalves10, 2021-09-23:
@Jaeyoung-Lim but if i want to use attitude setpoints, for instance, the quarterion coordinates for the fixed wing flr south is suposed to be x=0 y=0 z=0,7071 w=-0,7071. But when i send this coordinates the drone does not respond as shown in the video.

https://user-images.githubusercontent.com/89699717/134483883-3ac972f1-f64d-401c-8897-ddb9c2402e3f.mp4

joaoalves10, 2021-09-23:
@Jaeyoung-Lim but if i want to use attitude setpoints, for instance, the quarterion coordinates for the fixed wing flr south is suposed to be x=0 y=0 z=0,7071 w=-0,7071. But when i send this coordinates the drone does not respond as shown in the video.

https://user-images.githubusercontent.com/89699717/134484485-ab29704a-344b-4438-a53b-8b2b5fcf81fd.mp4

Then i tested what should be the quarterion coordinates when the aircraft is going south through a echo command, and the result was:

https://user-images.githubusercontent.com/89699717/134484617-e0f039e0-bcbe-440e-9075-e893887cc123.mp4

which means that the corrdinates that i sent in the first video should be correct.

Jaeyoung-Lim, 2021-09-23:
@joaoalves10 You are sending setpoints for offboard mode, but you are flying in mission mode.

joaoalves10, 2021-09-23:
@Jaeyoung-Lim that is in the second video, because i wanted to know which cordinates were related to north, west, east and south.
In the first video i was in offboard mode with the coordinates that i told you above.

Jaeyoung-Lim, 2021-09-23:
@joaoalves10 have you tried changing the setpoints?

joaoalves10, 2021-09-23:
So i first found the quarterion coordinates through the following command: rostopic echo /mavros/setpoint_raw/attitude_target. I put my aircraft flying heading north, west, east and south and note those coordinates, those are:

north: (x,y,z,w) = (0,0,-0.7071,-0,7071 )
west: (0,0,-1,0)
south: (0,0,0,7071, -0,7071)
east: (0,0,0,1)

after knowing this coordinates i went back to this message and substitute them in the orientation, then turn to OFFBOARD mode:

rostopic pub -r 5 /mavros/setpoint_raw/attitude mavros_msgs/AttitudeTarget "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
type_mask: 0b00000111
orientation: {x: 0.0, y: 0.0, z: -0.7071, w: -0,7071} - north
body_rate: {x: 0.0, y: 0.0, z: 0.0}
thrust: 0.5"

So after testing all those four coordinates the aircraft just did not respond. In all those coordinates the aircraft just mantain the heading that was flying or if the aircraft was loitering, it just stop turning and stated to follow a random heading.

I thought that if i put in orientation parameter, for instance, the north quarterion coordinates it would turn north, but that does not happen.

joaoalves10, 2021-09-23:
@Jaeyoung-Lim for instance, in this video i put the north coordinates but it does not work

https://user-images.githubusercontent.com/89699717/134573283-966ce7d0-636d-4ac2-87b6-a16d410717c8.mp4

Jaeyoung-Lim, 2021-09-24:
@Jaeyoung-Lim Okay, then it's probably something to do with you using rostopic pub.

I suggest that you publish the topics properly from a package. If you need an example you can look at https://github.com/Jaeyoung-Lim/mavros_controllers/blob/567d9d4771b2ffb38464765e040c31e0c982a5e9/geometric_controller/src/geometric_controller.cpp#L74

joaoalves10, 2021-09-24:
@Jaeyoung-Lim thank you so much for your help

I have already tried roll rate command but i would prefer the orientation.

One more thing, do you know if 
rostopic pub -r 5 /mavros/mount_control/mmand mavros_msgs/MountControl 
"header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
mode: 0
pitch: 0.0
roll: 0.0
yaw: 0.5
altitude: 0.0
latitude: 0.0
longitude: 0.0"

and

rostopic pub -r 5 /mavros/hil/controls mavros_msgs/HilControls
"header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
roll_ailerons: 0.0
pitch_elevator: 0.0
yaw_rudder: 0.0
throttle: 0.0
aux1: 0.0
aux2: 0.0
aux3: 0.0
aux4: 0.0
mode: 0
nav_mode: 0"

can do heading command. I say this because they have roll_ailerons, etc. My issue is to know what to put in de "mode2 and in "nav_mode".

if you know something i would appreciate, let me know. 

And thank you one more time for your help, it has been really helpful for my project development

Jaeyoung-Lim, 2021-09-24:
@joaoalves10 As I mentioned two times above, there is no command available for commanding "heading" 

If you can control roll angles(not roll rates), this means everything in the data pipeline is correct but the fw attitude controller does not take yaw orientation as a reference. Is this the case?

joaoalves10, 2021-09-24:
@Jaeyoung-Lim i tried to

> So i first found the quarterion coordinates through the following command: rostopic echo /mavros/setpoint_raw/attitude_target. I put my aircraft flying heading north, west, east and south and note [...]

@Jaeyoung-Lim i tried this in a quadcopter and it worked fine. I do not know why it does not work in a fixed wing...

joaoalves10, 2021-09-24:
> 
> 
> @joaoalves10 As I mentioned two times above, there is no command available for commanding "heading"
> 
> If you can control roll angles(not roll rates), this means everything in the data pipeline is correct but the fw attitude controller does not take yaw orientation as a reference. Is this the case?

@Jaeyoung-Lim My goal for the project is to have a fixed wing chasing a car. Through a onboard camera it will locate the car and estimate its coordinates and then will process it onboard. 
After having those coordinates, the onboard computer will control the UAV to follow the car. 
Thats why i need heading/roll control, to enable the onboard computer to send commands to guide the UAV according to the car position. 
Do you know if it is possible?

Jaeyoung-Lim, 2021-09-24:
> @Jaeyoung-Lim i tried this in a quadcopter and it worked fine. I do not know why it does not work in a fixed wing...

Fixedwing is using a different postion / attitude controller. All your attitude setpoints are orientations rotated around the z axis. I was asking you if you tested anything regarding the roll axis. If this works it means that your message is fine, but the fixedwing controller does not use the yaw orientation in your attitude setpoints. 

> My goal for the project is to have a fixed wing chasing a car. Through a onboard camera it will locate the car and estimate its coordinates and then will process it onboard.
After having those coordinates, the onboard computer will control the UAV to follow the car.
Thats why i need heading/roll control, to enable the onboard computer to send commands to guide the UAV according to the car position.
Do you know if it is possible?

All the autopilot code is public, so of course it is possible.

But as I mentioned(now the third time), sending attitude setpoints without any flight control is not enough. For example if you have a level attitude with arbitrary thrust setpoints the vehicle will not be able to maintain attitude. Therefore you either have to write your flight controller to run offboard or you need to modify the fixedwing attitude controller yourself.

joaoalves10, 2021-09-24:
@Jaeyoung-Lim i did not try to do it in the roll axis, do you know the coordinates?

Jaeyoung-Lim, 2021-09-24:
@joaoalves10 You can specify the quaternions to command any arbitrary roll commands. If you want to understand how quaternions work, you can read https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation

joaoalves10, 2021-09-24:
> All the autopilot code is public, so of course it is possible.
>
> But as I mentioned(now the third time), sending attitude setpoints without any flight control is not enough. For example if you have a [...]

@Jaeyoung-Lim which means that i will need to write the code by myself ou ajust the existing one?

joaoalves10, 2021-09-24:
> 
> 
> @joaoalves10 You can specify the quaternions to command any arbitrary roll commands. If you want to understand how quaternions work, you can read https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation

Thank you

joaoalves10, 2021-09-24:
> Fixedwing is using a different postion / attitude controller. All your attitude setpoints are orientations rotated around the z axis. I was asking you if you tested anything regarding the roll axis. [...]
@Jaeyoung-Lim i tried and i was able to send attitude coordinates for de x axis (roll), however, it is not coupled with the logitudinal axis (velocity and altitude), because it starts to climb or descent depending on the thrust parameter.
Do you know if it is possible to put a specific roll and the aircraft maintain the altitude during turn? or it will always depend on my thrust input?

Jaeyoung-Lim, 2021-09-24:
@joaoalves10 I am closing this thread since this is not a "issue" related to this repo. Please post questions on discuss.px4.io if you have mroe questions.

closed 2021-09-24 by Jaeyoung-Lim

joaoalves10, 2021-09-24:
thank you for your help