PX4/PX4-SITL_gazebo-classic #808: Why timeConstantUp is smaller than timeConstantDown

robin-shaun, 2021-09-26:
Hello, in iris.sdf.jinja, the timeConstantUp is smaller than timeConstantDown, which is strange for the BLDC model.
`      <timeConstantUp>0.0125</timeConstantUp>
      <timeConstantDown>0.025</timeConstantDown>`
 I think the  timeConstantUp should be larger than timeConstantDown.

Jaeyoung-Lim, 2021-09-26:
@robin-shaun Could you elaborate? Usually the time to reach a certain speed is faster, compared to slowing down (without breaking).

In either case, I don't think that parameter is currently being used.

miku39s3, 2021-10-06:
@Jaeyoung-Lim I have simulated the motor-propeller model with nonlinear aerodynamic torque in the MATLAB and the result shows that the acceleration response is indeed faster than the deceleration response. So using different time constant in two processes is reasonable. If the aerodynamic torque is linear, the acceleration and deceleration responses will be same.

closed 2021-10-07 by robin-shaun