PX4/PX4-SITL_gazebo-classic #823: development/mavlink.h Missing?

iamrajee, 2021-11-14:
While building on ubuntu20.04, I'm getting below error:
```
/home/rajendra/px4_ws/src/PX4-SITL_gazebo/include/gazebo_gimbal_controller_plugin.hh:39:10: fatal error: development/mavlink.h: No such file or directory
   39 | #include <development/mavlink.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
I've tried to built by both make -j$(nproc) -l$(nproc) and catkin_make, I get the same error.

I'm `origin/master` branch.

Logs
rajendra@u20:~/px4_ws/src/PX4-SITL_gazebo/build$ cmake ..
-- install-prefix: /usr
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
-- Found ccd: /usr/include (found suitable version "2.0", minimum required is "2.0") 
-- Found fcl: /opt/ros/noetic/include (found suitable version "0.6.1", minimum required is "0.3.2") 
-- Found assimp: /usr/include (found version "5.0.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
[... 223 lines cut ...]
[ 75%] Building CXX object CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o
[ 76%] Linking CXX shared library libForceVisual.so
[ 76%] Built target ForceVisual
[ 77%] Linking CXX shared library libgazebo_video_stream_widget.so
[ 77%] Built target gazebo_video_stream_widget
[ 78%] Linking CXX shared library libgazebo_wind_plugin.so
[ 78%] Built target gazebo_wind_plugin
[ 79%] Linking CXX shared library libgazebo_uuv_plugin.so
[ 79%] Built target gazebo_uuv_plugin
[ 80%] Linking CXX shared library libgazebo_drop_plugin.so
[ 81%] Linking CXX shared library libgazebo_sonar_plugin.so
[ 82%] Linking CXX shared library libgazebo_vision_plugin.so
[ 82%] Built target gazebo_drop_plugin
[ 82%] Built target gazebo_sonar_plugin
[ 83%] Linking CXX shared library libgazebo_controller_interface.so
[ 83%] Built target gazebo_vision_plugin
[ 83%] Built target gazebo_controller_interface
[ 84%] Linking CXX shared library libgazebo_opticalflow_plugin.so
[ 84%] Linking CXX shared library libgazebo_gst_camera_plugin.so
[ 84%] Built target gazebo_opticalflow_plugin
[ 84%] Built target gazebo_gst_camera_plugin
[ 85%] Linking CXX shared library libgazebo_user_camera_plugin.so
[ 85%] Built target gazebo_user_camera_plugin
make: *** [Makefile:171: all] Error 2
rajendra@u20:~/px4_ws/src/PX4-SITL_gazebo/build$

Jaeyoung-Lim, 2021-11-14:
@iamrajee You need a newer mavlink version that has the develop dialect

iamrajee, 2021-11-14:
> @iamrajee You need a newer mavlink version that has the develop dialect

You are right. After rolling back to previous version (git checkout 6fa6ec78a7a1619b7b90ac0c01aaec919defbe35), I no longer get above error.

Jaeyoung-Lim, 2021-11-14:
@iamrajee Not sure if you understood what I meant. The problem is that the development dialect is being used in this repo after https://github.com/PX4/PX4-SITL_gazebo/pull/822

However, your problem is that you don't have a development dialect in your mavlink package (not this repo)

closed 2021-11-15 by Jaeyoung-Lim

AnushaDevi-Embedded, 2022-01-09:
@iamrajee 

Hello, 

I am new to Mavlink and PX4. I am also facing the same issue. I did not understand the solution. Can you please elaborate?

wuyanze0226, 2024-10-21:
Hello. I meeting the same problem as you,would you mind telling me how to fix it? Thank you very much.