PX4/PX4-SITL_gazebo-classic #824: Compilation error:  too few arguments to function "uint16_t mavlink_msg_storage_information_pack_chan"

JohnVorwald, 2021-11-15:
The following install generates an error.

```
mkdir -p ~/src
cd ~/src
git clone --recursive https://github.com/PX4/sitl_gazebo.git
cd ~/src/sitl_gazebo
mkdir build
cd build
cmake ..
# make -j$(nproc) -l$(nproc)
make VERBOSE=1

CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o -c /home/john/src/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp
/home/john/src/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp: In member function ‘void gazebo::CameraManagerPlugin::_handle_storage_info(const mavlink_message_t*, sockaddr*)’:
/home/john/src/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp:778:5: error: too few arguments to function ‘uint16_t mavlink_msg_storage_information_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint32_t, uint8_t, uint8_t, uint8_t, float, float, float, float, float, uint8_t, const char*, uint8_t)’
     );
     ^
In file included from /opt/ros/melodic/include/mavlink/v2.0/common/common.h:2380:0,
                 from /opt/ros/melodic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/john/src/sitl_gazebo/include/gazebo_camera_manager_plugin.h:21,
                 from /home/john/src/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp:18:
/opt/ros/melodic/include/mavlink/v2.0/common/./mavlink_msg_storage_information.h:159:24: note: declared here
 static inline uint16_t mavlink_msg_storage_information_pack_chan(uint8_t system_id, uint8_t component_id, uint8_t chan,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
CMakeFiles/gazebo_camera_manager_plugin.dir/build.make:62: recipe for target 'CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o' failed
make[2]: *** [CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o] Error 1
make[2]: Leaving directory '/home/john/src/sitl_gazebo/build'
CMakeFiles/Makefile2:276: recipe for target 'CMakeFiles/gazebo_camera_manager_plugin.dir/all' failed
make[1]: *** [CMakeFiles/gazebo_camera_manager_plugin.dir/all] Error 2
make[1]: Leaving directory '/home/john/src/sitl_gazebo/build'
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
Suggestions on correcting.

Jaeyoung-Lim, 2021-11-15:
@JohnVorwald You have an outdated mavlink version. Have you tried updating your mavlink?

JohnVorwald, 2021-11-15:
I'm new.  I followed the video "Ardupilot & PX4 SIL (SITL) from 0 to 100 in one Hour (All in One)".  How do I update mavlink.  I started down the following path, but I don't know how to fill in the gui prompts.

```
pip3 install --user future
sudo apt install python3-lxml libxml2-utils
sudo apt install python3-tk
cd ~/src
git clone https://github.com/mavlink/mavlink.git –recursive
PYTHONPATH=~/src/mavlink/
python3 -m mavgenerate
```

Jaeyoung-Lim, 2021-11-15:
@JohnVorwald What you want is the mavlink c libraries: 
```
      run: git clone --depth 1 https://github.com/mavlink/c_library_v2.git /usr/local/include/mavlink && rm -rf /usr/local/include/mavlink/.git

```

closed 2021-11-20 by Jaeyoung-Lim