PX4/PX4-SITL_gazebo-classic #829: 'make px4_sitl_default gazebo' gives 'ninja: build stopped: subcommand failed. make: *** [Makefile:225: px4_sitl_default] Error 1'

joelfinnigan, 2021-11-18:
I am very inexperienced with the Linux console, and SITL simulation in general. I am running Ubuntu Linux 20.04 on Parallels, attempting to build a PX4 SITL simulation. I began with a fresh virtual machine and installed ROS Noetic using the following instructions: https://wiki.ros.org/noetic/Installation/Ubuntu

Running 'roscore & rosrun gazebo_ros gazebo' launches both the server and the client correctly, and the proper ROS connections and Gazebo services are set up as described here: http://gazebosim.org/tutorials?tut=ros_installing&cat=connect_ros

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ rostopic list
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/performance_metrics
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ rosservice list
/gazebo/apply_body_wrench
/gazebo/apply_joint_effort
/gazebo/clear_body_wrenches
/gazebo/clear_joint_forces
/gazebo/delete_light
/gazebo/delete_model
/gazebo/get_joint_properties
/gazebo/get_light_properties
/gazebo/get_link_properties
/gazebo/get_link_state
/gazebo/get_loggers
/gazebo/get_model_properties
/gazebo/get_model_state
/gazebo/get_physics_properties
/gazebo/get_world_properties
/gazebo/pause_physics
/gazebo/reset_simulation
/gazebo/reset_world
/gazebo/set_joint_properties
/gazebo/set_light_properties
/gazebo/set_link_properties
/gazebo/set_link_state
/gazebo/set_logger_level
/gazebo/set_model_configuration
/gazebo/set_model_state
/gazebo/set_parameters
/gazebo/set_physics_properties
/gazebo/spawn_sdf_model
/gazebo/spawn_urdf_model
/gazebo/unpause_physics
/rosout/get_loggers
/rosout/set_logger_level

I then installed the PX4 autopilot by running 'git clone https://github.com/PX4/PX4-Autopilot.git --recursive' and then 'bash ./PX4-Autopilot/Tools/setup/ubuntu.sh'

Once complete, I ran 'cd ~/PX4-Autopilot' then 'make clean' then 'make px4_sitl_default gazebo'

The build did not work, giving the following output:

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ make px4_sitl_default gazebo
-- PX4 version: v1.13.0-alpha1-3017-g17ad7071c3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/parallels/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
[... 220 lines cut ...]
    linked by target "gazebo_user_camera_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_camera_manager_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "LiftDragPlugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[825/833] Building CXX object platforms/posix/CMakeFiles/px4.dir/apps.cpp.o
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/parallels/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/parallels/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[827/833] Building CXX object src/modu...es__mavlink.dir/mavlink_messages.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:225: px4_sitl_default] Error 1

Please help!!

Jaeyoung-Lim, 2021-11-18:
@joelfinnigan You can see the error in the log:
```
1.0'
-- No package 'gstreamer-app-1.0' found
```

joelfinnigan, 2021-11-19:
Hi, I tried to install gstreamer using instructions from this post: https://github.com/PX4/PX4-Autopilot/issues/13416#issuecomment-552136098

I restarted and attempted to build again however I unfortunately still got an error related to gstreamer:

> parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ make px4_sitl_default gazebo
> -- PX4 version: v1.13.0-alpha1-3455-g43fb076828
> -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
> -- PX4 config file: /home/parallels/PX4-Autopilot/boards/px4/sitl/default.px4board
> -- PLATFORM posix
> -- ROMFSROOT px4fmu_common
> -- TESTING y
> -- ETHERNET y
> -- PX4 config: px4_sitl_default
> -- PX4 platform: posix
> -- PX4 lockstep: enabled
> -- cmake build type: RelWithDebInfo
> -- The CXX compiler identification is GNU 9.3.0
> -- The C compiler identification is GNU 9.3.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /usr/bin/cc
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Building for code coverage
> -- ccache enabled (export CCACHE_DISABLE=1 to disable)
> -- ROMFS: ROMFS/px4fmu_common
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/parallels/PX4-Autopilot/build/px4_sitl_default
> [7/828] git submodule src/drivers/gps/devices
> [9/828] git submodule src/modules/mavlink/mavlink
> [235/828] git submodule Tools/sitl_gazebo
> [501/828] Generating Mavlink developme...ssage_definitions/v1.0/development.xml
> Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
> Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
> Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
> Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
> Validating /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Parsing /home/parallels/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
> Merged enum MAV_CMD
> Merged enum MAV_CMD
> Found 222 MAVLink message types in 4 XML files
> Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_sitl_default/mavlink/development
> Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_sitl_default/mavlink/standard
> Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_sitl_default/mavlink/common
> Generating C implementation in directory /home/parallels/PX4-Autopilot/build/px4_sitl_default/mavlink/minimal
> Copying fixed headers for protocol 2.0 to /home/parallels/PX4-Autopilot/build/px4_sitl_default/mavlink
> [667/828] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- cmake build type: RelWithDebInfo
> -- The C compiler identification is GNU 9.3.0
> -- The CXX compiler identification is GNU 9.3.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
[... 139 lines cut ...]
>     linked by target "gazebo_imu_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_mavlink_interface" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_motor_model" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_camera_manager_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_magnetometer_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_barometer_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "LiftDragPlugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
>     linked by target "gazebo_catapult_plugin" in directory /home/parallels/PX4-Autopilot/Tools/sitl_gazebo
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
> See also "/home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
> [822/828] Linking CXX executable bin/px4
> FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
> cd /home/parallels/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/parallels/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/parallels/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
> [824/828] Linking CXX executable bin/px4
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:225: px4_sitl_default] Error 1
> 

I then tried 'sudo apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev' before attempting to build again, however this now causes terminal to crash:

[image]

Any ideas?

joelfinnigan, 2021-11-19:
Have tried reinstalling streamer with:

> sudo apt-get install --reinstall gstreamer1.0-alsa gstreamer1.0-libav gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-pulseaudio libgstreamer-plugins-bad1.0-0 libgstreamer-plugins-base1.0-0 libgstreamer-plugins-good1.0-0 libgstreamer1.0-0

However building still causes terminal crash - seems to crash on the same step each time (39/130 whilst generating typhoon_h480)

Jaeyoung-Lim, 2021-11-19:
@joelfinnigan Did the error change after installing thr gstreamer plugins?

joelfinnigan, 2021-11-19:
@Jaeyoung-Lim yes, before I was just getting the error message but now it causes terminal to freeze and crash?

Jaeyoung-Lim, 2021-11-19:
@joelfinnigan How much memory do you have on your system? if it is less than 8GB, you are probably running out of memory

joelfinnigan, 2021-11-20:
@Jaeyoung-Lim thanks you were right! Increased the memory allocation of the VM and build is now working and the GUI opens! Am now getting a new error message when I try to initiate takeoff however:

> _____  __   __    ___ 
> | ___ \ \ \ / /   /   |
> | |_/ /  \ V /   / /| |
> |  __/   /   \  / /_| |
> | |     / /^\ \ \___  |
> \_|     \/   \/     |_/
> 
> px4 starting.
> 
> INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
> INFO  [init] found model autostart file as SYS_AUTOSTART=10016
> INFO  [param] selected parameter default file eeprom/parameters_10016
> [param] parameter file not found, creating eeprom/parameters_10016
>   SYS_AUTOCONFIG: curr: 0 -> new: 1
>   SYS_AUTOSTART: curr: 0 -> new: 10016
>   CAL_ACC0_ID: curr: 0 -> new: 1310988
>   CAL_GYRO0_ID: curr: 0 -> new: 1310988
>   CAL_ACC1_ID: curr: 0 -> new: 1310996
>   CAL_GYRO1_ID: curr: 0 -> new: 1310996
>   CAL_ACC2_ID: curr: 0 -> new: 1311004
>   CAL_GYRO2_ID: curr: 0 -> new: 1311004
>   CAL_MAG0_ID: curr: 0 -> new: 197388
>   CAL_MAG1_ID: curr: 0 -> new: 197644
>   SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
>   SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
> * SYS_AUTOCONFIG: curr: 1 -> new: 0
>   IMU_INTEG_RATE: curr: 200 -> new: 250
> INFO  [dataman] data manager file './dataman' size is 11798680 bytes
> PX4 SIM HOST: localhost
> INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
> INFO  [simulator] Simulator connected on TCP port 4560.
> INFO  [commander] LED: open /dev/led0 failed (22)
> Gazebo multi-robot simulator, version 11.9.0
> Copyright (C) 2012 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org
> 
> [Msg] Waiting for master.
>   PWM_AUX_OUT: curr: 0 -> new: 1234
>   PWM_MAIN_OUT: curr: 0 -> new: 1234
> [Msg] Connected to gazebo master @ http://127.0.0.1:11345
> [Msg] Publicized address: 10.211.55.9
> INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
> INFO  [init] setting PWM_AUX_OUT none
> INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
> INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
> INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
> INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
> INFO  [logger] logger started (mode=all)
> INFO  [logger] Start file log (type: full)
> INFO  [logger] [logger] ./log/2021-11-20/13_02_52.ulg	
> INFO  [logger] Opened full log file: ./log/2021-11-20/13_02_52.ulg
> INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
> INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
> INFO  [px4] Startup script returned successfully
> pxh> INFO  [tone_alarm] home set
> INFO  [tone_alarm] notify negative
> pxh> commander takeoff
> pxh> INFO  [commander] Armed by internal command	
> INFO  [navigator] Using minimum takeoff altitude: 2.50 m	
> INFO  [commander] Takeoff detected	
> WARN  [commander] Failsafe enabled: no datalink	
> INFO  [commander] Failsafe mode activated	
> INFO  [tone_alarm] battery warning (fast)
> INFO  [navigator] RTL HOME activated	
> INFO  [navigator] RTL: landing at home position.	
> INFO  [navigator] RTL: climb to 489 m (1 m above destination)	
> INFO  [commander] Failsafe mode deactivated	
> INFO  [navigator] RTL: return at 489 m (1 m above destination)	
> INFO  [navigator] RTL: land at destination	
> INFO  [commander] Landing detected	
> INFO  [commander] Disarmed by landing	
> INFO  [logger] closed logfile, bytes written: 1364578

I think this may be because I don't have QGroundControl installed? I have tried installing using the instructions given here: https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html

However I get this error:

> parallels@ubuntu-linux-20-04-desktop:~$ ~/Downloads/QGroundControl.AppImage
> bash: /home/parallels/Downloads/QGroundControl.AppImage: cannot execute binary file: Exec format error

Is this because I have an ARM chipset or have I missed a step somewhere? If it is the former, what would you recommend?

Jaeyoung-Lim, 2021-11-20:
@joelfinnigan It looks like it, but not sure.

joelfinnigan, 2021-11-20:
@Jaeyoung-Lim what would you recommend?

Jaeyoung-Lim, 2021-11-20:
@joelfinnigan I would try finding an arm image of QGC. In any case, your issue is not related to simulation anymore.

closed 2021-11-20 by Jaeyoung-Lim

joelfinnigan, 2021-11-22:
@Jaeyoung-Lim ok thanks