PX4/PX4-SITL_gazebo-classic #833: sitl_multiple_vehicle plane spwans inverted

airsimdevdd, 2021-12-11:
[image]

Jaeyoung-Lim, 2021-12-13:
@airsimdevdd Please explain how you can reproduce this problem exactly.

Currently I am a bit limited on bandwidth but in the meantime if you can fix the script that would be great: https://github.com/PX4/PX4-Autopilot/blob/master/Tools/gazebo_sitl_multiple_run.sh

airsimdevdd, 2021-12-13:
I'm using latest PX4 and gazebo version and script  command -"Tools/gazebo_sitl_multiple_run.sh -m plane -n 2"

Jaeyoung-Lim, 2021-12-13:
@airsimdevdd Which gazebo version?

airsimdevdd, 2021-12-13:
gazebo 9.19.0-1~bionic

airsimdevdd, 2021-12-13:
@Jaeyoung-Lim sir could  you help me out when u are free

airsimdevdd, 2021-12-15:
problem solved using older px4 version thanks for help @Jaeyoung-Lim

Jaeyoung-Lim, 2022-01-22:
@

Jaeyoung-Lim, 2022-01-22:
@airsimdevdd Using an older version of PX4 is not fixing the problem

airsimdevdd, 2022-01-24:
yes there is something issue in script gazebo_sitl_multiple_run.sh

airsimdevdd, 2022-01-24:
In the script  i changed line 42  to "gz model --spawn-file=/tmp/${MODEL}_${N}.sdf --model-name=${MODEL}_${N} -x 0.0 -y $((3*${N})) -z 1.0 "     works fine

Jaeyoung-Lim, 2022-01-24:
@airsimdevdd I have fixed the issue in https://github.com/PX4/PX4-Autopilot/pull/19063 Thanks!

closed 2022-01-24 by Jaeyoung-Lim