PX4/PX4-SITL_gazebo-classic #834: How can we use multple offboard programs to communicate with the simulated fcu?

tadteo, 2021-12-11:
Hi,
I am trying to simulate the connection (bidirectional) with a drone with two different computers in SITL. 
Essentially, I have an onboard computer that is connected with the fcu using mavros (as default in the mavros simulation), and in addition, a would like to simulate another computer/program that uses pymavlink. 

For the part using mavros there is no problem, it listens to port 14540 and connects to the fcu. However, I am not able to connect with another program to the fcu without interfering. 
I have tried to use some other port 14541-14549 but they are not broadcasting the fcu messages. Moreover, I don't think I would be able to send messages to the fcu since, the fcu is listening just on port 14580.

The only way I have found to do that is to use the connection for the GCS to connect my program to the drone but this is not the best choice since I would still like to use QGroundControl. 

I was thinking to use mavlink-router on the GCS side to be able to use multiple programs/devices, but it is not the cleanest solution.

What is the best way to configure this kind of communication in PX4-SITL?

Jaeyoung-Lim, 2021-12-11:
@tadteo How are you running the multivehicle simulation? It should just work with the instructions in https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html

You just need to be careful with setting mavros ports.

If you are using ROS(since you are using mavros) you can use this script: https://github.com/Jaeyoung-Lim/mavros_humantracking/blob/master/humantracking_controller/launch/multi_sitl_humantrack_circle.launch as reference

tadteo, 2021-12-11:
hi @Jaeyoung-Lim,

No, I'm not running a multivehicle simulation. Actually, the second program runs on a simulation of a landing station that needs to communicate with the drone, but it does not run the PX4 and it is neither a GCS. It is something different.

In the multivehicle case, it generates multiple fcu instances. In my case, I just have one single fcu instance but multiple programs that need to communicate with the same fcu.

tadteo, 2021-12-11:
[image]

This is a graphical explanation starting from the px4 sitl general overview. (Sorry for the poor quality drawing)

Jaeyoung-Lim, 2021-12-11:
@tadteo Ah sorry for overlooking your question.

The camera manager plugin: https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_camera_manager_plugin.cpp is already doing this and it is using `udp:14030`

If you want to see which ports are avaialble maybe look at the rcS script: https://github.com/PX4/PX4-Autopilot/blob/0f6e30599c4933deb7f39f357c4981c1f17f0ac8/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L8

tadteo, 2021-12-11:
Ok thank you very much! I will try to do that by modifying those files. And will get back when done.

tadteo, 2021-12-16:
Hi,

I have solved this by modifying the rcS script and adding new ports and the link.

These are the lines that I have added to [this file](https://github.com/PX4/PX4-Autopilot/blob/0f6e30599c4933deb7f39f357c4981c1f17f0ac8/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L8):
```
udp_extra_port_local=$((18280+px4_instance))
udp_extra_port_remote=$((18270+px4_instance))

#Extra port link
mavlink start -x -u $udp_extra_port_local -r 4000000 -f -m onboard -o $udp_extra_port_remote
```

Jaeyoung-Lim, 2022-01-19:
@tadteo Thanks for the update

closed 2022-01-19 by Jaeyoung-Lim