PX4/PX4-SITL_gazebo-classic #835: Physics_msgs does not have a name type

aliimranTW, 2021-12-23:
Hi, 

I am trying to use the `gazbo_wind_plugin` with my ROS/Gazebo package. The purpose is to simulate a custom drone in a simulation with wind in it. I have copied the cpp and .h file to my package, however, I get the following error while `catkin_make`:

> /home/mowaistahir/test3_ws/src/src/sjtu-drone/include/gazebo_wind_plugin.h:131:3: error: ‘physics_msgs’ does not name a type
   physics_msgs::msgs::Wind wind_msg;
   ^~~~~~
In file included from /home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:22:0:
/home/mowaistahir/test3_ws/src/src/sjtu-drone/include/common.h:47:26: error: redefinition of ‘const string gazebo::kDefaultNamespace’
 static const std::string kDefaultNamespace = "";
                          ^~~~~~~
In file included from /home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:21:0:
/home/mowaistahir/test3_ws/src/src/sjtu-drone/include/gazebo_wind_plugin.h:36:26: note: ‘const string gazebo::kDefaultNamespace’ previously declared here
 static const std::string kDefaultNamespace = "";
                          ^~~~~~~
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp: In member function ‘virtual void gazebo::GazeboWindPlugin::Load(gazebo::physics::WorldPtr, sdf::ElementPtr)’:
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:91:39: error: ‘physics_msgs’ was not declared in this scope
   wind_pub_ = node_handle_->Advertise<physics_msgs::msgs::Wind>("~/" + wind_pub_topic_, 10);
                                       ^~~~~~
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:91:29: error: parse error in template argument list
   wind_pub_ = node_handle_->Advertise<physics_msgs::msgs::Wind>("~/" + wind_pub_topic_, 10);
                             ^~~~~~~~~~~~~
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:91:91: error: no matching function for call to ‘gazebo::transport::Node::Advertise<<expression error> >(std::__cxx11::basic_string<char>, int)’
 handle_->Advertise<physics_msgs::msgs::Wind>("~/" + wind_pub_topic_, 10);
                                                                        ^
In file included from /usr/include/gazebo-9/gazebo/transport/transport.hh:6:0,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:21,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/mowaistahir/test3_ws/src/src/sjtu-drone/include/gazebo_wind_plugin.h:29,
                 from /home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:21:
/usr/include/gazebo-9/gazebo/transport/Node.hh:153:31: note: candidate: template<class M> gazebo::transport::PublisherPtr gazebo::transport::Node::Advertise(const string&, unsigned int, double)
       transport::PublisherPtr Advertise(const std::string &_topic,
                               ^~~~~
/usr/include/gazebo-9/gazebo/transport/Node.hh:153:31: note:   template argument deduction/substitution failed:
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:91:91: error: template argument 1 is invalid
 handle_->Advertise<physics_msgs::msgs::Wind>("~/" + wind_pub_topic_, 10);
                                                                        ^
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp: In member function ‘void gazebo::GazeboWindPlugin::OnUpdate(const gazebo::common::UpdateInfo&)’:
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:145:3: error: ‘wind_msg’ was not declared in this scope
   wind_msg.set_frame_id(frame_id_);
   ^~~~
/home/mowaistahir/test3_ws/src/src/sjtu-drone/src/gazebo_wind_plugin.cpp:145:3: note: suggested alternative: ‘sendmmsg’
   wind_msg.set_frame_id(frame_id_);
   ^~~~
   sendmmsg
src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/build.make:62: recipe for target 'src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/src/gazebo_wind_plugin.cpp.o' failed
make[3]: * [src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/src/gazebo_wind_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:1325: recipe for target 'src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/all' failed
make[2]: * [src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/all] Error 2
CMakeFiles/Makefile2:1337: recipe for target 'src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/rule' failed
make[1]: * [src/sjtu-drone/CMakeFiles/gazebo_wind_plugin.dir/rule] Error 2
Makefile:630: recipe for target 'gazebo_wind_plugin' failed
make: * [gazebo_wind_plugin] Error 2
Invoking "make clean gazebo_wind_plugin -j4 -l4" failed

How do I resolve this `physics_msgs` issue?

Jaeyoung-Lim, 2021-12-23:
@aliimranTW You can check how the physics_msgs are defined in yhe CMakeLists.txt : https://github.com/PX4/PX4-SITL_gazebo/blob/27298574ce33a79ba6cfc31ed4604974605e7257/CMakeLists.txt#L314

aliimranTW, 2021-12-23:
@Jaeyoung-Lim Thank you for your response.
I included these lines in my CMakeLists.txt:

> set(physics_msgs
    msgs/Force.proto
    msgs/Wind.proto
    )
PROTOBUF_GENERATE_CPP(PHY_PROTO_SRCS PHY_PROTO_HDRS ${physics_msgs})
add_library(physics_msgs SHARED ${PHY_PROTO_SRCS})
link_libraries(physics_msgs)

The error persists even after that. Is there something else needed to be done in addition to adding these lines of code?

Jaeyoung-Lim, 2021-12-26:
@aliimranTW I dont think copy pasting would do much.  Better try and understand the CMakelists is doing and adapt it to your project

closed 2022-01-21 by Jaeyoung-Lim