PX4/PX4-SITL_gazebo-classic #836: How to run HITL with simulator connected via UDP to flight controller ?

MatCapLynx, 2022-01-07:
Hi,
I'm triying to run HITL simulation with a pixhawk connected via USB to a companion computer. The problem is that this computer doesn't have a GUI. To run my simulator (Gazebo), I have to use another computer on the same network. Mavros runs on the companion computer.

This is exactly the case described on [this](https://docs.px4.io/master/en/simulation/hitl.html#:~:text=PX4%20and%20QGroundControl.-,Note,-The%20simulator%20can) note but I hav'nt been able to set it up.

I think this can be configured in my iris_hitl.sdf file :
```xml
<plugin name='mavlink_interface' filename='libgazebo_mavlink_interface.so'>
      <robotNamespace/>
      <imuSubTopic>/imu</imuSubTopic>
      <magSubTopic>/mag</magSubTopic>
      <baroSubTopic>/baro</baroSubTopic>
      <mavlink_addr>INADDR_ANY</mavlink_addr>
      <mavlink_tcp_port>4560</mavlink_tcp_port>
      <mavlink_udp_port>14560</mavlink_udp_port>
      <serialEnabled>0</serialEnabled>
      <serialDevice>/dev/ttyACM0</serialDevice>
      <baudRate>921600</baudRate>
      <qgc_addr>INADDR_ANY</qgc_addr>
      <qgc_udp_port>14550</qgc_udp_port>
      <sdk_addr>INADDR_ANY</sdk_addr>
      <sdk_udp_port>14540</sdk_udp_port>
      <hil_mode>1</hil_mode>
      <hil_state_level>0</hil_state_level>
      <send_vision_estimation>0</send_vision_estimation>
      <send_odometry>1</send_odometry>
      <enable_lockstep>0</enable_lockstep>
      <use_tcp>0</use_tcp>
```
and also by setting the right `fcu_url` and `gcu_url` when launching mavros. I've tried a lot of different configurations without success. What is the correct network configuration for my case ?

Jaeyoung-Lim, 2022-01-19:
@MatCapLynx Why do you need mavros for the  HITL simulation?

MatCapLynx, 2022-01-21:
I need it to get/send data to the Pixhawk. I have specific sensors that cannot be simulated but that are needed to run my tests.