PX4/PX4-SITL_gazebo-classic #837: how to make a custom urdf for multiple uav in gazebo

airsimdevdd, 2022-01-21:
can one guide me for creating a custom urdf file for multiple uav in gazebo @Jaeyoung-Lim

Jaeyoung-Lim, 2022-01-21:
@airsimdevdd You can have a look at the models in this repo and how they are templated using [jinja](https://jinja.palletsprojects.com/en/3.0.x/) templates: https://github.com/PX4/PX4-SITL_gazebo/blob/master/models/iris/iris.sdf.jinja

The scripts are generated in https://github.com/PX4/PX4-Autopilot/blob/bcd057ac3eb77ca4f1afef532a74843ada1bbe18/Tools/gazebo_sitl_multiple_run.sh#L38

airsimdevdd, 2022-01-21:
I cannot use 
jinja files because jinja files are used by the px4 v1.12.3  and there is problem with it there are spawning inverted and crashed image shown below  ,but 1.11.3 use xacro file which generates the .urdf files at a temp location and spawns good
[image]

Jaeyoung-Lim, 2022-01-21:
@airsimdevdd You can change the spawn location via the script: https://github.com/PX4/PX4-Autopilot/blob/bcd057ac3eb77ca4f1afef532a74843ada1bbe18/Tools/gazebo_sitl_multiple_run.sh#L42

If you want to use urdf files, then you are free to do so :)

airsimdevdd, 2022-01-24:
getting error while creating a custom sdf model gazbeo doesnot open mentioned in new issue #838

closed 2022-01-28 by Jaeyoung-Lim