PX4/PX4-SITL_gazebo-classic #844: Can't take-off my custom model

airsimdevdd, 2022-02-03:
I imported my custom model which is similar to delta_wing it does not take off just glides is there any problem with lift drag plugin any one would help me in this . I'm new to this

Jaeyoung-Lim, 2022-02-08:
@airsimdevdd Have you compared your model with the plane model? You can get hints by comparing your model with working models to find out what problems you have with your custom model.

airsimdevdd, 2022-02-08:
yes i checked it  everything seems ok i think its unable to produce enough lift like delta-wing we have only elevons

Jaeyoung-Lim, 2022-02-08:
@airsimdevdd If you have figured out that you are not able to produce enough lift, you can either decrease your mass or increase the lift coefficients

Jaeyoung-Lim, 2022-02-21:
Closing as this is a support question, and there was no response 

if you have more questions feel free to reopen or ask on the public forum discuss.px4.io

closed 2022-02-21 by Jaeyoung-Lim