PX4/PX4-SITL_gazebo-classic #845: Wind speed in windy.world doesn't work!

iamrajee, 2022-02-21:
I'm running px4-sitl gazebo with windy.world. I did not change anything inside in world file.  When I take off the drone using QGC, there seems to be no effect to be on the drone. I have also tried to give wind speed in gazebo UI after opening and enable the wind for the model as well but still no luck.

[Here](https://youtu.be/fa1SYOLxop8) is the short video of the issue. Would really appreciate any help :)

Jaeyoung-Lim, 2022-02-21:
@iamrajee I think there is something wrong on yourside when starting the sim. Have you tried without using ROS?

```
make px4_sitl gazebo_iris__windy
```

iamrajee, 2022-02-21:
@Jaeyoung-Lim  Thank you so much for your input. I just tried without ROS as you suggest, I guess it did work(drone fly in slant orientation).

How can I run it with ROS?

I have cloned `PX4-SITL_gazebo` and `PX4-Autopilot` in my `ROS_Workspace/src` and have built using the `catkin_make`. Could that be the reason why I couldn't run with ROS? Do I have to clone `PX4-Autopilot` outside the ROS workspace and then source accordingly?

Thanks in advance for your help.

Jaeyoung-Lim, 2022-02-21:
@iamrajee I would first check if the windy.world is being spawned

iamrajee, 2022-02-21:
True, world looks different in both cases.

Gazebo windy.world  when launched using ROS
`roslaunch px4 posix_sitl.launch world:=windy.world`
[image]

Gazebo windy.world  when launched without ROS
`make px4_sitl gazebo_iris__windy`
[image]

Moreover, windy.world file is exact the same in both cases.

Jaeyoung-Lim, 2022-02-21:
You need to pass the full path of the world fie, not just the name of the world file.

iamrajee, 2022-02-21:
Thank you for your prompt response, after giving the full absolute path, it worked. Sorry for the confusion.

closed 2022-02-21 by iamrajee