PX4/PX4-SITL_gazebo-classic #846: ROS2 Offboard Mode Only Works with The First Vehicle

adk4davsys, 2022-02-22:
Hello,

I am trying to run multi-vehicle SITL simulation and noticed that any ROS2 publishers from the second vehicle gets ignored by the gazebo. The weird thing about it is that setpoints seem to be working. Arming and changing flight mode are the two things I have noticed to be getting ignored by gazebo. I also observed via `ros2 topic echo` to see that I am publishing to the correct topic (the first vehicle arms and changes flight mode according to my publishers).

Currently the setup is running PX4 1.12.3 with gazebo version 11.10.1.

Please let me know if you need any further information. Thank you!

Jaeyoung-Lim, 2022-02-22:
@adk4davsys Could you explain how this is related to gazebo? How are you sending arming commands using ROS?

adk4davsys, 2022-02-23:
Hi @Jaeyoung-Lim, thank you for your reply. At this point I am only starting to investigate so I do not know if this is strictly related to gazebo. But some odd behaviors during SITL simulation made me suspect the issue stems from gazebo side.

By utilizing ROS2, I am sending arming commands to two SITL multicopters via the topic `vehicle_command` with the following:
```
set_mode_msg.timestamp = gps_time;
set_mode_msg.param1 = 1; // from px4_msgs/msgs/vehicle_command
set_mode_msg.param2 = 0; // 0: arm-disarm unless prevented by safety checks (i.e. when landed)
set_mode_msg.command = px4_msgs::msg::VehicleCommand::VEHICLE_CMD_COMPONENT_ARM_DISARM;
set_mode_msg.target_system = 1;
set_mode_msg.target_component = 1;
set_mode_msg.source_system = 1;
set_mode_msg.source_component = 1;
set_mode_msg.from_external = true;

set_mode_pub->publish(set_mode_msg);
```
The first vehicle arms without an issue along with accepting other commands such as `AUTO.LAND`. The second vehicle on the other hand, ignores arming command. I have to manually arm the second vehicle via QGC. Weird thing is the second vehicle does not ignore `trajectory_setpoint` topic and flies the waypoints that I send using ROS2.

Any help or hints on getting this debugged will be very helpful. In the meantime I will try with three or four vehicles to see what the behaviors are like.

Thank you for your time.

adk4davsys, 2022-02-23:
@Jaeyoung-Lim Here is how I start my SITL:
```
./Tools/gazebo_sitl_multiple_run.sh -t px4_sitl_rtps -m iris -n 2
```
Starting two agents on separate terminals:
```
$ micrortps_agent -t UDP -r 2020 -s 2019 -n ac_0 &
$ micrortps_agent -t UDP -r 2022 -s 2021 -n ac_1 &
```
Then I start my ROS2 nodes which include the arming function described above. I am separating the topics with node specific namespace to publish to: `ac_0/fmu/vehicle_command/in` or `/ac_1/fmu/vehicle_command/in`.

closed 2022-03-07 by adk4davsys