PX4/PX4-SITL_gazebo-classic #848: VTOL drop model fails to takeoff for airspeed sensor

AhmedElsafy, 2022-02-24:
I am able to do simulations using the standard VTOL, but the VTOL drop model is suffering from an airspeed sensor alarm 

`INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.	
`
I have edited the sdf file to add a link and joint for an airspeed sensor and it took off but then the VTOL crashes when transitioning to FW mode 

thank you

Jaeyoung-Lim, 2022-02-28:
@AhmedElsafy Can you share the full log?

AhmedElsafy, 2022-02-28:
@Jaeyoung-Lim can you please guide me where to pull that log from please ?

AhmedElsafy, 2022-03-02:
`[0______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=1043
INFO  [param] selected parameter default file eeprom/parameters_1043
[param] parameter file not found, creating eeprom/parameters_1043
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 1043
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
Gazebo multi-robot simulator, version 10.2.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.82.43.106
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
* MAV_TYPE: curr: 2 -> new: 22
etc/init.d-posix/rcS: 21: [: !=: unexpected operator
INFO  [init] Mixer: etc/mixers-sitl/standard_vtol_sitl.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-03-02/14_24_20.ulg	
INFO  [logger] Opened full log file: ./log/2022-03-02/14_24_20.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> [Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
INFO  [commander] Failsafe mode activated	
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.	
INFO  [commander] Failsafe mode deactivated	
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
WARN  [PreFlightCheck] Preflight Fail: Airspeed invalid.
INFO  [tone_alarm] notify negative

`

AhmedElsafy, 2022-03-09:
@Jaeyoung-Lim Is that the right info ?

sfuhrer, 2022-03-22:
It seems like there is no airspeed being simulated in the vtol_drop (`make px4_sitl gazebo_standard_vtol_drop`). 
You can check that by typing "listener airspeed" in the console.