PX4/PX4-SITL_gazebo-classic #855: Startup script returns errors with return value:256

yhdurkaya, 2022-03-22:
Hello folks,

I have created a an iris drone with r200. I have followed model creation guideline and I believe it is ok.
https://gazebosim.org/tutorials?tut=build_model

 Then I changed the sitl_run.sh to use my drone instead of classic iris. Now, when building with make px4_sitl gazebo, it throws some errors saying
"ERROR [init] Unknown model iris_r200 (not found by name on /home/USER/catkin_ws/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/airframes)
ERROR [px4] Startup script returned with return value: 256 ".

I have also added a new airframe configuration in "/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/airframes" following this page.
"https://docs.px4.io/master/en/dev_airframes/adding_a_new_frame.html"

Everytime I try to build, the same error throws and I do not have a single idea to fix it. Anyone encountered with this before?
Thanks in advance.

Jaeyoung-Lim, 2022-03-22:
It means your model is not on the GAZEBO_MODEL_PATH. Therefore there is something wrong on either how you created the model or how your system is configured.

Also, you have the firmware file inside a catkin workspace. This can be a source of a lot of problems and hoghly discouraged

yhdurkaya, 2022-03-22:
It actually is on my GAZEBO_MODEL_PATH, I just checked bashrc. I put my model files into the models folder in sitl_gazebo and when I open a fresh gazebo, I can see and insert my model without any problem.

Thanks for the advice about workspace.

Jaeyoung-Lim, 2022-03-22:
@yhdurkaya Ah sorry, I now see the error
```
"ERROR [init] Unknown model iris_r200 (not found by name on /home/USER/catkin_ws/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/airframes)
ERROR [px4] Startup script returned with return value: 256 ".
```

This means that you are missing an airframe config for your vehicle. For example, you need a vehicle named `iris_r200` in this directory: https://github.com/PX4/PX4-Autopilot/tree/master/ROMFS/px4fmu_common/init.d-posix/airframes and also add it to the CMakeLists.txt

yhdurkaya, 2022-03-22:
I have created a file just like you said and added it to CMakeLists.txt. 

```
#!/bin/sh
#
# @name 3DR Iris Quadrotor with r200 camera
#
# @type Quadrotor Wide
# @class Copter
#
# @output MAIN1 motor 1
# @output MAIN2 motor 2
# @output MAIN3 motor 3
# @output MAIN4 motor 4
#
# @output AUX1 feed-through of RC AUX1 channel
# @output AUX2 feed-through of RC AUX2 channel
# @output AUX3 feed-through of RC AUX3 channel
# @output AUX4 feed-through of RC FLAPS channel
#
# @maintainer Lorenz Meier <lorenz@px4.io>
#
# @board px4_fmu-v2 exclude
# @board bitcraze_crazyflie exclude
#

. ${R}etc/init.d/rc.mc_defaults

# TODO tune roll/pitch separately
param set-default MC_ROLL_P 7
param set-default MC_ROLLRATE_I 0.05
param set-default MC_ROLLRATE_D 0.004
param set-default MC_PITCH_P 7
param set-default MC_PITCHRATE_I 0.05
param set-default MC_PITCHRATE_D 0.004
param set-default MC_YAW_P 2.5
param set-default MC_YAWRATE_P 0.25
param set-default MC_YAWRATE_I 0.25

param set-default BAT1_V_DIV 12.27559
param set-default BAT1_A_PER_V 15.391030303103

set MIXER quad_w

```

This is the file that I've added and I basically copied the Iris. My current problem is that it says 
```
ERROR [param] Parameter BAT1_V_DIV not found.
ERROR [px4] Startup script returned with return value: 65280
```

But BAT1_V_DIV is right there, it doesn't throw errors about MC_ROLL_P or anything else but somehow it can't find this parameter.

Thanks for the previous problem, couldn't have done it if it is not for your help.