PX4/PX4-SITL_gazebo-classic #859: gzserver: symbol lookup error

O2KAN, 2022-04-06:
gzserver: symbol lookup error: ~/git/simulation/.gitman/px4_firmware/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so: undefined symbol: _ZN8mav_msgs4msgs17CommandMotorSpeedC1Ev

I have a symbolic lookup error. I couldn't find a solution to fix it. With this error, I can not open the simulation on gazebo correctly.

Jaeyoung-Lim, 2022-04-06:
@O2KAN How are you building this part of the code?

O2KAN, 2022-04-06:
I am taking this error for any sitl simulation. make px4_sitl_gazebo,  roslaunch mavros_posix_sitl.launch, roslaunch geometric_controller sitl_trajectory_track_circle.launch. I think you are talking about these lines for building.

[image]
[image]
g

Jaeyoung-Lim, 2022-04-06:
@O2KAN Try cleaning up your message definitions. (Note that the following commands will remove all your uncommited changes)
```
make distclean
make submodulesupdate
make px4_sitl gazebo
```

O2KAN, 2022-04-06:
They did not work. I deleted and install everything again. Now the workspace is not building. 
[image]
[image]

O2KAN, 2022-04-06:
I am getting this error while installing a repository. In that repository px4, mavlink atc. are built-in. When I start the installation they automatically install. 
I applied the same commands you said, it did not work. Later I decided to delete the repository and installed it again. This error is what I get now. 

Thank you for your helping

O2KAN, 2022-04-16:
This issue is still unresolved. Do you have any idea? 
[image]
[image]

Jaeyoung-Lim, 2022-04-16:
@O2KAN please share the full log

Your log you shared shows a different issue with the original issue you have posted

O2KAN, 2022-04-16:
Actually, they are the same issues, gzserver: symbolic lookup error, just the screenshots are taken from different workspaces.

Jaeyoung-Lim, 2022-04-16:
@O2KAN please fully explain which workspace is running what. You shouldnt need two workspaces since everything is under the firmware repo.

The error is caused by the faxt that you have two copies of this repo in two separate workspaces

O2KAN, 2022-04-21:
Yes you are right I don't need two same px4 firmware repo. I installed the other one maybe somehow I can fix the error. 

I don't know but the error was fixed when I tried to use jmsim and install it.  Now working fine. Thank you for your help.

closed 2022-04-21 by O2KAN