PX4/PX4-SITL_gazebo-classic #864: Mavlink Interfacing with Gazebo

huiyulhy, 2022-05-23:
Hi, I'm creating a custom vehicle in SITL to interface with Gazebo for my simulations. Using the same code snippet from Gazebo as the jinja gazebo_plane (but without control surfaces as I just want to test the plugins), the px4-rc.simulator file does not get run (and mavlink interface doesn't start in SITL), but it runs just fine in the gazebo_plane model. 

Comparing the command line outputs between the gazebo_plane SITL and the custom SITL, I can see that TCP port 4560 is not connected for the SITL of the custom vehicle, and is probably the reason why there isn't a SITL mavlink interface. This was the only file change (and link to a custom airframe), hence I was wondering why the TCP port connection was not picked up. 

[image]

![custom vehicle command line output ](https://user-images.githubusercontent.co
[image]
m/26198479/169880231-afbf40cc-687b-4c57-a858-bf70d845f552.png)

Any advice is greatly appreciated!

huiyulhy, 2022-05-24:
@Jaeyoung-Lim I copied the parameters (exactly from gazebo_plane) into the custom vehicle airframe config, and tried running the same custom vehicle SITL. This was the full output in the terminal: 

[image]
[image]

The first part of the output is the same for the custom airframe, but the parts that differ is shown in the post above

Jaeyoung-Lim, 2022-05-24:
@huiyulhy The initial screenshot shared looked lile the pareters were not loadong for the custom airframe. Do you have an imu and a mavlink interface in your custom airframe config? Can you do a diff between yours and the plane on the mavlink interface plugin?

huiyulhy, 2022-05-26:
@Jaeyoung-Lim I realize the issue had not been with mavlink interface. that works fine, turns out that it had been a definition error on one of my links that was preventing this from working. Thanks for helping to debug!

closed 2022-05-26 by huiyulhy