PX4/PX4-SITL_gazebo-classic #865: How to use libgazebo_motor_model.so plugin in my project

jfaure6, 2022-05-25:
Hello Github.

After compiling and installing the libs and plugins on my computer, I want to use it in my project, especially the libgazebo_motor_model.so plugin. How can I integrate it in my project ? Maybe by editing my CmakeLists.txt ? I'm a beginner in CMake...

Thanks in advance.

huiyulhy, 2022-05-26:
you'll need to call the plugin in your sdf.jinja file for your model, and point to the libgazebo_motor_model.so plugin. An example of calling the plugin in the sdf.jinja file can be found in the gazebo_plane iris model. Refer to Tools -> sitl_gazebo -> models -> plane -> plane.sdf.jinja

jfaure6, 2022-05-31:
Thank you for your answer ! I managed to put the libgazebo_motor_model.so in my project. Attched to this message, there are the important files used to spawn the px4_vision (xacro extension in fact, but txt to import here).

It spawns well, but when I launch my flocking service for my project, nothing happens and my rotors and drones don't take off.

Do you have an idea of what to do ? 
[base.txt](https://github.com/PX4/PX4-SITL_gazebo/files/8804259/base.txt)
[spawn_px4_vision.txt](https://github.com/PX4/PX4-SITL_gazebo/files/8804267/spawn_px4_vision.txt)
[body_macro.txt](https://github.com/PX4/PX4-SITL_gazebo/files/8804271/body_macro.txt)
[motor_macro.txt](https://github.com/PX4/PX4-SITL_gazebo/files/8804273/motor_macro.txt)