PX4/PX4-SITL_gazebo-classic #867: Cant't spwan model to gazebo

liumou5, 2022-06-02:
I try to use two gazebo in one roscore, so I renamed the node of one gazebo
```
liu@liu-System-Product-Name:~/tools/PX4-Autopilot$ rosnode list
/gazebo0
/gazebo_gui
/rosout
/uav0/mavros
/uav0/vehicle_spawn_liu_System_Product_Name_6746_6598916509689035951
/uav1/mavros
/uav1/vehicle_spawn_liu_System_Product_Name_6746_1389833989942486500
/uav2/mavros
/uav2/vehicle_spawn_liu_System_Product_Name_6746_3255675674634546547
```
and gazebo service's namespaces also changed:
```
liu@liu-System-Product-Name:~/tools/PX4-Autopilot$ rosservice list |grep "gazebo"
/gazebo0/apply_body_wrench
/gazebo0/apply_joint_effort
/gazebo0/clear_body_wrenches
/gazebo0/clear_joint_forces
/gazebo0/delete_light
/gazebo0/delete_model
/gazebo0/get_joint_properties
/gazebo0/get_light_properties
/gazebo0/get_link_properties
/gazebo0/get_link_state
/gazebo0/get_loggers
/gazebo0/get_model_properties
/gazebo0/get_model_state
/gazebo0/get_physics_properties
/gazebo0/get_world_properties
/gazebo0/pause_physics
/gazebo0/reset_simulation
/gazebo0/reset_world
/gazebo0/set_joint_properties
/gazebo0/set_light_properties
/gazebo0/set_link_properties
/gazebo0/set_link_state
/gazebo0/set_logger_level
/gazebo0/set_model_configuration
/gazebo0/set_model_state
/gazebo0/set_parameters
/gazebo0/set_physics_properties
/gazebo0/spawn_sdf_model
/gazebo0/spawn_urdf_model
/gazebo0/unpause_physics
/gazebo_gui/get_loggers
/gazebo_gui/set_logger_level
```
now it seems like that px4 can't link to gazebo service, and it stops at here: 
```
[INFO] [1654131999.272022, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol0
[INFO] [1654131999.281703, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1654131999.379047, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol1
[INFO] [1654131999.390562, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1654131999.493954, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol2
[INFO] [1654131999.503091, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1654131999.921183443, 0.004000000]: waitForService: Service [/gazebo0/set_physics_properties] is now available.
[ INFO] [1654131999.930937625, 0.012000000]: Physics dynamic reconfigure ready.

```

so i serched `"spawn_sdf_model"` in file `PX4-Autopilot` and found this file:
```
liu@liu-System-Product-Name:~/tools/PX4-Autopilot$ grep -r "spawn_sdf_model" .
./integrationtests/python_src/px4_it/util/flight_path_assertion.py:        self.spawn_model = rospy.ServiceProxy('/gazebo/spawn_sdf_model', SpawnModel)
./integrationtests/python_src/px4_it/util/flight_path_assertion.py:        #self.spawn_model = rospy.ServiceProxy('/gazebo0/spawn_sdf_model', SpawnModel)
```
and changed some code in `PX4-Autopilot/integrationtests/python_src/px4_it/util/flight_path_assertion.py`, like this:
```
     def __init__(self, positions, tunnelRadius=1, yaw_offset=0.2):
         threading.Thread.__init__(self)
         rospy.Subscriber("vehicle_local_position", vehicle_local_position, self.position_callback)
         #self.spawn_model = rospy.ServiceProxy('/gazebo/spawn_sdf_model', SpawnModel)                
         #self.set_model_state = rospy.ServiceProxy('/gazebo/set_model_state', SetModelState)
         #self.delete_model = rospy.ServiceProxy('/gazebo/delete_model', DeleteModel)
 
         self.spawn_model = rospy.ServiceProxy('/gazebo0/spawn_sdf_model', SpawnModel)
         self.set_model_state = rospy.ServiceProxy('/gazebo0/set_model_state', SetModelState)
         self.delete_model = rospy.ServiceProxy('/gazebo0/delete_model', DeleteModel)
```
but px4 stil try to link to `/gazebo/spawn_sdf_model` instead of `/gazebo0/spawn_sdf_model`. 
```
[INFO] [1654132065.867992, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol0
[INFO] [1654132065.877111, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1654132065.992833, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol1
[INFO] [1654132066.001965, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1654132066.138037, 0.000000]: Loading model XML from ros parameter sdf_standard_vtol2
[INFO] [1654132066.147238, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1654132066.311244678, 0.004000000]: waitForService: Service [/gazebo0/set_physics_properties] is now available.
[ INFO] [1654132066.321165001, 0.016000000]: Physics dynamic reconfigure ready.
```
Do you know how to solve this problem?

Jaeyoung-Lim, 2022-06-02:
@liumou5 Could you elaborate on why yo had to modify the default multivehicle script? This should already be functional for your use case

liumou5, 2022-06-02:
> @liumou5 Could you elaborate on why yo had to modify the default multivehicle script? This should already be functional for your use case

Thanks for your reply. 
We have to simulate more than fifty UAV at the same time, but the real time factor will drop from 1 to about 0.3, so we try to use two computers or just two Gazebo in one computer. 
it's OK, we found a solution, and I'll close this issue soon

liumou5, 2022-06-02:
we found a solution.
px4 use file `single_vehicle_spawn.launch` to spawn model in Gazebo. and this fill called `spawn_model` in ros package `gazebo_ros`. and `spawn_model` use argument "gazebo_namespace" to determine gazebo namespace, so we just need add that when call this fill

closed 2022-06-02 by liumou5