PX4/PX4-SITL_gazebo-classic #874: fatal error: development/mavlink.h

De-chuan, 2022-06-27:
When i use make -j(nproc) -l(nproc),i met this question,

> h:39:10: fatal error: development/mavlink.h: 没有那个文件或目录
   39 | #include <development/mavlink.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Jaeyoung-Lim, 2022-06-27:
@De-chuan Please follow the documentation. You are missing mavlink in ypur system

De-chuan, 2022-06-27:
> @De-chuan Please follow the documentation. You are missing mavlink in ypur system

I find mavlink.h in Firmware/build/px4_sitl_default/mavlink/develompent.But i make -j(nproc) -l(nproc) that also has the question.

Jaeyoung-Lim, 2022-06-29:
@De-chuan The make target you are showing is not really a supported build target since this package only considers a standalone within catkin_ws

If you want to support the make target you need to manually link the mavlink library

spurnvoj, 2022-08-22:
Hi @Jaeyoung-Lim, as you probably know, the `development` dialect is available with release `mavlink 2022.8.8-1`. However, `#include <development/mavlink.h>` still doesn't exist because the header file is now available under a different path `#include <mavlink/v2.0/development/mavlink.h>`. 
Are you planning to make a hotfix for the version [48440d7](https://github.com/PX4/PX4-SITL_gazebo/tree/48440d7b5c78a21182415266334981f1163f4b2c) of the `SITL_gazebo` that is locked with the stable release `v1.13.0` of the `PX4-Autopilot`?

Jaeyoung-Lim, 2022-08-22:
Fixed in #896

closed 2022-08-22 by Jaeyoung-Lim