PX4/PX4-SITL_gazebo-classic #876: Gazebo message size for Odometry not correct

ecmnet, 2022-07-12:
In the header of the odometer message the size of the message is specified. Adding 'repeating' fields for covariance according to the `odometer.proto` definition, increases the size of the message. However, as the same instance is reused in the publishing loop, covariances are added, instead of replaced. 

A simple solution to this would be to clear the covariance fields of the message before re-using them, like

`odom_msg.clear_velocity_covariance();
  odom_msg.clear_pose_covariance();`

before filling up the message again.

Jaeyoung-Lim, 2022-07-12:
@ecmnet Would you be able to create a PR to address this?

ecmnet, 2022-07-13:
@Jaeyoung-Lim Here you are.

Jaeyoung-Lim, 2022-08-24:
Fixed in #878

closed 2022-08-24 by Jaeyoung-Lim